petrobras/ross #784: Error in response display peak-peak?

LeifParr, 2021-03-12:
I tried to scale response results in peak peak, as was suggested and fixed earlier, but I couldn't get it to work.
I ran a forced response with the new private method:
_calculate_major_axis_per_node(node, 'major', amplitude_units= 'pkpk_m')
This returned results equal to:
_calculate_major_axis_per_node(node, 'major', amplitude_units= 'm')
Also, what I wanted to do was:
_calculate_major_axis_per_node(node, 'major', amplitude_units= 'pkpk_mm')
but this crashed:
pint.errors.UndefinedUnitError: 'pkpk_mm' is not defined in the unit registry

Has anyone else tested the peak-peak results function? For now I manually multiply by two, so it isn't a super high priority fix.

LeifParr, 2021-03-12:
On the same note, I couldn't find the pkpk option mentioned in the documentation or doc string for the non-private response plotting methods?

raphaeltimbo, 2021-03-12:
Hi @LeifParr !

Ideally you should call the `plot_magnitude()` method in this case with something like:
```python
import ross as rs

rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
force = rotor._unbalance_force(3, 10.0, 0.0, speed)
resp = rotor.run_forced_response(force=force, speed_range=speed)
resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='pkpk_m')
```
I looked little into the code, but I believe we have some pre-processing within `plot_magnitude` before calling `_calculate_major_axis_per_node`, so I think it may be a problem to call it directly.
Just out of curiosity, were you testing something or did you feel that the non-private method is missing some option and you needed to go with the private one?

Regarding the `pkpk_mm`, you are right. I will look at it.

LeifParr, 2021-03-12:
I am grateful for swift replies from the Ross team!
I agree with your notion here of using the public plot_magnitude function.
However, my goal was to extract the response values as numpy arrays for further processing. To this end, I see that I can extract the series data using the plotly figures:
   resp_fig = response.plot_magnitude(probes)
   resp_speeds = resp_fig['data'][0]['x']
   vib_data = np.zeros((len(probes),len(resp_speeds)))
   for i, probe in enumerate(probes):
        # here was the other method, using the private function:
        #node = probe[0]
        #major_res = response._calculate_major_axis_per_node(node, 'major', amplitude_units= 'm')
        #vib_data[i,:] = np.abs(major_res[3,:]) # take absolute value
        vib_data[i,:]=resp_fig['data'][i]['y']

Before closing the case, I also noticed that the documentation does not mention the possibility to use 'major' and 'minor' for the probe angle in these plots?

raphaeltimbo, 2021-03-12:
> However, my goal was to extract the response values as numpy arrays for further processing. To this end, I see that I can extract the series data using the plotly figures:

Interesting! 
The purpose of these `Results` object is to provide the user with plots and also with all the data that is used to make these plots.
The vibration data is actually available in the `response.forced_response` attribute. The problem is that there you have raw data in complex values.
I think we could create some intermediate methods that will generate the data before creating the figures, and we could make those methods available to the user.
Something like:
```python
import ross as rs

rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
force = rotor._unbalance_force(3, 10.0, 0.0, speed)
resp = rotor.run_forced_response(force=force, speed_range=speed)
# if you want only the data:
resp.data_magnitude(probe=[(0, 'major')], amplitude_units='pkpk_m')
# if you want the plot
resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='pkpk_m')
```

> Before closing the case, I also noticed that the documentation does not mention the possibility to use 'major' and 'minor' for the probe angle in these plots?

Yes, it is missing. Going to work on that.

Also, regarding the `amplitude_units='pkpk_cm'`, it seems that it is a limitation with the parsing capabilities of pint.
I have opened an issue there to check: https://github.com/hgrecco/pint/issues/1264

The following will work for now: `resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='pkpk_centimeter')`

LeifParr, 2021-03-12:
Excellent idea for the data magnitude. 
The reason I want data, and not only figures, is for applications like the one I am trying to work my way through now. 

We made a field balancing test on a 100 MW hydro power unit last month. The measurements provide a relation between a known applied unbalance at the generator rotor, and a measured vibration response in the bearings - both shaft vibration and housing vibration. What I am about to do is to iterate through a range of values for the stiffness of the bearing oil film and housing in ROSS, and see if I can come close to the measured vibration. This will serve as verification of the calculated properties of bearing and housing.

raphaeltimbo, 2021-03-12:
Nice to hear that ROSS is helping you in that!
Latter I am planning on using the 'Discussions' session here in the repository to post some use cases that we have here at work.

Regarding this issue, I think we could leave it open to link with the PR that I am going to open to fix docs and to add the `data` method.

rodrigomoliveira1, 2021-03-12:
Hello @raphaeltimbo and @LeifParr 

Doing some quick work based on what you guys are talking, I came to this one below:

```python
    def data_magnitude(
        self,
        probe,
        probe_units="rad",
        frequency_units="rad/s",
        amplitude_units="m",
    ):
        """Return the forced response (magnitude) in array format.

        Parameters
        ----------
        probe : list of tuples
            List with tuples (node, orientation angle, tag).
            node : int
                indicate the node where the probe is located.
            orientation : float
                probe orientation angle about the shaft. The 0 refers to +X direction.
            tag : str, optional
                probe tag to be displayed at the legend.
        probe_units : str, option
            Units for probe orientation.
            Default is "rad".
        frequency_units : str, optional
            Units for the x axis.
            Default is "rad/s"
        amplitude_units : str, optional
            Units for the y axis.
            Acceptable units dimensionality are:
                '[length]' - Displays the displacement;
                '[speed]' - Displays the velocity;
                '[acceleration]' - Displays the acceleration.
            Default is "m" 0 to peak.
            To use peak to peak use the prefix 'pkpk_' (e.g. pkpk_m)

        Returns
        -------
        data : dictionary
            Dictionary storing magnitude data arrays. They keys are set based on the
            probe's tag.
        """
        frequency_range = Q_(self.speed_range, "rad/s").to(frequency_units).m

        unit_type = str(Q_(1, amplitude_units).dimensionality)
        try:
            base_unit = self.default_units[unit_type][0]
        except KeyError:
            raise ValueError(
                "Not supported unit. Dimensionality options are '[length]', '[speed]', '[acceleration]'"
            )

        data = {}
        data["frequency_range"] = frequency_range

        for i, p in enumerate(probe):
            angle = Q_(p[1], probe_units).to("rad").m
            vector = self._calculate_major_axis_per_node(
                node=p[0], angle=angle, amplitude_units=amplitude_units
            )[3]
            try:
                probe_tag = p[2]
            except IndexError:
                probe_tag = f"Probe {i+1} - Node {p[0]}"

            data[probe_tag] = Q_(np.abs(vector), base_unit).to(amplitude_units).m

        return data
```

Thus, here is an example how it would work:

```python
>>> import ross as rs

>>> rotor = rs.rotor_example()
>>> speed = np.linspace(0, 1000, 101)
>>> force = rotor._unbalance_force(3, 10.0, 0.0, speed)
>>> resp = rotor.run_forced_response(force=force, speed_range=speed)
>>> # if you want only the data:
>>> data = resp.data_magnitude(probe=[(0, 'major', 'probe1'), ], amplitude_units='pkpk_m')

>>> data["probe1"]
array([...])
```

raphaeltimbo, 2021-03-16:
That is looking good @rodrigomoliveira1. 
One thing that I would like to check is if using a namedtuple instead of a dictionary would be good for the user.
The advantage in this case would be using `data.probe1` instead of `data["probe1"] for example. 
In this case it is easier to type and (in some cases) easier to have autocomplete.
So it is a minor advantage and we would still have to look for the disadvantages.
What do you guys think?

Also, regarding the use of pkpk, in the pint issue it was suggested that we could use a pkpk constant instead of a prefix.
In this case these would be valid code:
```python
resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='pkpk m') # no underscore

resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='m pkpk') # pkpk could go after the length unit

resp.plot_magnitude(probe=[(0, 'major')], amplitude_units='cm pkpk') # cm would already work in this case
```
So, although this will break some code, I think it is an improvement on what we have now.

LeifParr, 2021-03-16:
If I could venture an opinion, I think I would prefer a 2d Numpy array as output, with frequency range as col (or row) zero. Could also return the probes, since the tags may be updated.

data[i,:] = Q_(np.abs(vector), base_unit).to(amplitude_units).m
return data, probes

rodrigomoliveira1, 2021-03-17:
@LeifParr 

This way, it can be a little confusing to correlate the data with the respective probes, if the user uses several of them.
Another option I see is setting a Pandas Dataframe, where the frequency is the col zero and the probes follows.

rodrigomoliveira1, 2021-03-17:
Great! I will work on it and open a PR soon.

raphaeltimbo, 2021-03-18:
Sorry for not responding before, but I am very happy that you guys went with the dataframe solution. 
Results will still have labels to avoid confusion (as @rodrigomoliveira1 pointed) and we will have all 
the pandas features - easy to convert to arrays, dotted attributes, easy to export to csv, excel etc.
Very good solution!

commit 19a1868b (2021-04-12): Change pkpk from prefixe to constant

This change improves the syntax when using pkpk and solves the issues
that we were having when trying to use 'pkpk_cm' for example. Instead
of 'pkpk_m' now we can write 'm pkpk', without the underscore. Other
cases such as 'cm pkpk' will work in this case.

Closes #784.
closed 2021-04-12 by raphaeltimbo