petrobras/ross #786: trouble with plotting campbell diagram

GitDiscipline, 2021-04-12:
r = rs.rotor_example()
speed_range = np.linspace(0, 1000, 100)
campbell = r.run_campbell(speed_range)
campbell.plot(harmonics=[0.5, 1])

this is the code I have used according to the ROSS documentation. But after running the code, no plot is seen. There is syntax or semantic error.

raphaeltimbo, 2021-04-12:
Hi there @GitDiscipline !
Are you running your code in a jupyter notebook or in an IDE?

The issue #733 also has some discussion about this that could help you.

GitDiscipline, 2021-04-13:
Hi @raphaeltimbo 
      I  am using IDLE.  #733 has helped me.  Adding  .show()  resolved the issue
 
Thanks!

raphaeltimbo, 2021-04-13:
Great @GitDiscipline !
I will close this issue now. Feel free to reopen it if you have problems in the future.

closed 2021-04-13 by raphaeltimbo