petrobras/ross #814: Fix - 4 and 6 DoF EIG solution method modification

verg1lio, 2021-06-30:
Issue related to the numerical precision of the solver method adopted for the eigenvalue problem, aiming to change the solver for a more precise and robust one. The current solver works fine for small matrices (smaller than ~30x30), but struggles to keep the error acceptable for anything bigger than that. Some models demand bigger and more poorly conditioned matrices, which demand a different solver.

raphaeltimbo, 2021-06-30:
Hi there @verg1lio !
The function used to calculate eigenvalues basically has this in it: https://github.com/ross-rotordynamics/ross/blob/f7cc3d118a727313bf648805b1b5e72b610e3f6d/ross/rotor_assembly.py#L1237-L1251

So we are using the scipy [eigs](https://docs.scipy.org/doc/scipy/reference/generated/scipy.sparse.linalg.eigs.html) function, which uses the Fortran [ARPACK](https://www.caam.rice.edu/software/ARPACK/) to calculate the eigenvalues. 

> ARPACK is a collection of Fortran77 subroutines designed to solve large scale eigenvalue problems.

As you can see in the code we have to set some arguments for the calculation of the eigenvalues when calling the `la.eigs()` function. I think we should start this investigation by checking if any of these arguments are having a negative impact on the calculation precision. 

We also need to discuss what level of precision we want here, since changing the solver may affect calculation time and also can increase our dependencies, which could increase package installation size, time to build and test the software etc. So we will need to balance those aspects as well.

To evaluate this, I would recommend that we select some study cases (models) to use as reference when comparing the solvers in precision and calculation time. If you could share the models that you are seeing a larger error here we could start by those.

verg1lio, 2021-06-30:
Hello @raphaeltimbo !

I start by agreeing with you here  _"As you can see in the code we have to set some arguments for the calculation of the eigenvalues when calling the la.eigs() function."_ , since if this routine already uses sufficiently accurate enough eigenvalue solution methods it is a simple method of using the right parameters.

Also regarding the precision, the solver we generally use applies a method called Cholesky factorization, which gives the first "n" eigenvalues and associated eigenvectors, being either the largest or the smallest ones [.](https://www.mathworks.com/help/matlab/ref/chol.html) This is a quick and reliable method that we have been using for decades with no issues in our codes. If i got it right, the "mpmath" package has this methodology, and is already a requirement in ROSS [.](https://mpmath.org/doc/current/matrices.html#matrix-operations) Nevertheless, if the la.eigs() have this capability, then the problem has a trivial solution.

As for the case study I completely agree. I will generate the same problem we were having while doing the test functions and comparing the eigenvalue solutions for the 6 DoF model. Having the problem I will post it here.

Anyways, I will get back to you here, and link a pull request we made this morning in our repository [.](https://github.com/LMEst-Rotor/ross/tree/fix_eig_method) to this issue.

verg1lio, 2021-07-06:
Hi @raphaeltimbo , good afternoon!

I believe I have identified the problem. It seems that the solver have different methods depending on the size or complexity of the eigenvalue problem it is trying to solve. See the attached images for clarification.

This first one have 36 finite elements, and falls in the "big" category for the solver, which utilizes a better method and gives constant and reliable results. Check that the Wns are constantly the same.

[image]

This last one, which we used in the example, utilizes 6 finite elements and fall in the "small" category, using classical triangulation methods which have little precision and tend to accumulate errors exponentially with the matrix size and conditioning. The results are not stable as the bigger matrix case are, and fluctuate from one execution to the other.

[image]

What intrigued me though is that these results are much more precise then the ones we were obtaining when writing the test functions about a year ago. I know that using increasingly bigger matrices will decrease the simple methods precision, but what i do not know is the threshold between methodologies.

In any event, I strongly recommend that we standardize the solver and make it independent of matrix size, possibly by changing the routine options to set a fixed method for all matrix sizes.

(@elyqg , @hiagopinacio)

raphaeltimbo, 2021-07-08:
Hi @verg1lio ! Could you paste here the code used to generate the 6 elements model so I can copy it and make some tests?

verg1lio, 2021-07-08:
Yes, absolutely! I was using our example routine as a base, and it goes as below:

```
import ross as rs
import numpy as np
rotor6 = rs.rotor_assembly.rotor_example_6dof()
camp6 = rotor6.run_campbell(np.linspace(0,400,101),frequencies=18)
rotor_speed = 100.0 # rad/s
modal6 = rotor6.run_modal(rotor_speed)
print(f"Undamped natural frequencies:\n {modal6.wn}")
```

You will notice that there is a "n=6" parameter in the example (rotor_example_6dof, in rotor_assembly.py) that should be changed for a different number of elements in the model.

verg1lio, 2021-07-08:
@raphaeltimbo 

The complete code is the example itself, as follows:
```

    i_d = 0
    o_d = 0.05
    n = 36
    L = [0.25 for _ in range(n)]

    shaft_elem = [
        ShaftElement6DoF(
            material=steel,
            L=0.25,
            idl=0,
            odl=0.05,
            idr=0,
            odr=0.05,
            alpha=0,
            beta=0,
            rotary_inertia=False,
            shear_effects=False,
        )
        for l in L
    ]

    disk0 = DiskElement6DoF.from_geometry(
        n=2, material=steel, width=0.07, i_d=0.05, o_d=0.28
    )
    disk1 = DiskElement6DoF.from_geometry(
        n=4, material=steel, width=0.07, i_d=0.05, o_d=0.28
    )

    kxx = 1e6
    kyy = 0.8e6
    kzz = 1e5
    cxx = 0
    cyy = 0
    czz = 0
    bearing0 = BearingElement6DoF(
        n=0, kxx=kxx, kyy=kyy, cxx=cxx, cyy=cyy, kzz=kzz, czz=czz
    )
    bearing1 = BearingElement6DoF(
        n=6, kxx=kxx, kyy=kyy, cxx=cxx, cyy=cyy, kzz=kzz, czz=czz
    )

    rotor6(shaft_elem, [disk0, disk1], [bearing0, bearing1])

    rotor_speed = 100.0 # rad/s

    modal6 = rotor6.run_modal(rotor_speed)
    print(f"Undamped natural frequencies:\n {modal6.wn}")

```

raphaeltimbo, 2021-07-08:
Hi @verg1lio , thank you for providing the code!
I have carried out some tests and I will try to explain the results that I got and also give some options on what we can do to improve things.

We can see in the code that we have the `v0` argument for the `las.eigs()` function:
https://github.com/ross-rotordynamics/ross/blob/f7cc3d118a727313bf648805b1b5e72b610e3f6d/ross/rotor_assembly.py#L1237-L1251

This value is set to `None` when we instantiate the rotor, and if we call `la.eigs` with this, we have:
```
v0 : ndarray, optional
    Starting vector for iteration. Default: random
```
So, the default value for the starting vector is random, but after the first eigenvalue calculation we set this value to: 
```python
self._v0 = np.real(sum(evectors.T))
```
We set this value for v0 to improve the performance for the next calculations.

If we set this to a fixed value, we can have repeatability (but we loose speed as we will see):

```python
for i in range(5):
    rotor6._v0 = np.ones(len(rotor6.A()))
    modal6 = rotor6.run_modal(rotor_speed)
    print(f"Undamped natural frequencies:\n {modal6.wn}")

Undamped natural frequencies:
 [5.17964778e-05 3.53466379e+00 3.54867463e+00 2.24056139e+01
 2.25269639e+01 3.13659949e+01]
Undamped natural frequencies:
 [5.17964778e-05 3.53466379e+00 3.54867463e+00 2.24056139e+01
 2.25269639e+01 3.13659949e+01]
Undamped natural frequencies:
 [5.17964778e-05 3.53466379e+00 3.54867463e+00 2.24056139e+01
 2.25269639e+01 3.13659949e+01]
Undamped natural frequencies:
 [5.17964778e-05 3.53466379e+00 3.54867463e+00 2.24056139e+01
 2.25269639e+01 3.13659949e+01]
Undamped natural frequencies:
 [5.17964778e-05 3.53466379e+00 3.54867463e+00 2.24056139e+01
 2.25269639e+01 3.13659949e+01]
```
Another way to have repeatability is to use `la.eig()` instead of `la.eigs()`, which uses the eigenvalue routine from LAPACK instead of ARPACK:

 ```python
for i in range(5):
    modal6 = rotor6.run_modal(rotor_speed, sparse=False)
    print(f"Undamped natural frequencies:\n {modal6.wn}")

Undamped natural frequencies:
 [8.82902875e-05 3.53466380e+00 3.54867463e+00 2.24056144e+01
 2.25269652e+01 3.13660164e+01]
Undamped natural frequencies:
 [8.82902875e-05 3.53466380e+00 3.54867463e+00 2.24056144e+01
 2.25269652e+01 3.13660164e+01]
Undamped natural frequencies:
 [8.82902875e-05 3.53466380e+00 3.54867463e+00 2.24056144e+01
 2.25269652e+01 3.13660164e+01]
Undamped natural frequencies:
 [8.82902875e-05 3.53466380e+00 3.54867463e+00 2.24056144e+01
 2.25269652e+01 3.13660164e+01]
Undamped natural frequencies:
 [8.82902875e-05 3.53466380e+00 3.54867463e+00 2.24056144e+01
 2.25269652e+01 3.13660164e+01]
```

Now, if we carry out some performance comparisons, we have:

```python
# Current implementation with las.eigs()
%%timeit
for i in range(10):
    modal6 = rotor6.run_modal(rotor_speed)

435 ms ± 4.57 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

# Setting initial vector to constant value
%%timeit
for i in range(10):
    rotor6._v0 = np.ones(len(rotor6.A()))
    modal6 = rotor6.run_modal(rotor_speed)

692 ms ± 8.77 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

# Do not consider sparse, use la.eig()
%%timeit
for i in range(10):
    modal6 = rotor6.run_modal(rotor_speed, sparse=False)

1.1 s ± 8.81 ms per loop (mean ± std. dev. of 7 runs, 1 loop each)

```

So, with the current code we have a 2x increase in calculation speed. 
Regarding the repeatability, I agree that it is not expected to see these values changing between each calculation, so I think we could address this in the documentation, stating that the method that we use as default could lead to this behavior and showing the option to change to `sparse=False` more clearly.
Also, I think we could add a 'global' config parameter, so that the user could do something like:
```python
import ross as rs

rs.config.eigen_sparse = False
```
and this would be used in other parts of the code, instead of having to pass the argument `sparse=False` to each function (this is actually not possible to some run functions such as `run_campbell`)

Regarding the tests, we have two options:
- set a higher `rtol` within the test;
- set `sparse=False` when calling the calculation.

For cases that we have in the tests right now, I feel that we have tolerances that are acceptable within rotordynamics.

verg1lio, 2021-07-08:
Hi again @raphaeltimbo !

I got the same impression you have. For rotordynamics simulations the results are reasonably accurate and within tolerances, and the high error we were detecting when writing the test functions is not present as it was. I could not replicate that error when trying it this week, and believe it to have been patched by the library devs. 

In any event, the increase in computational time may prove problematic for a very detailed Campbell diagram generation, since it roughly doubled in time with no sensible gain in precision.

With that said, _exclusively_ for the test functions I believe that setting the `sparse=False` option is the way to go, since it will allow for an effective and reliable conference of the code. I say that for in my experience many additions to the physics of a model may yield almost negligible effects in the end results, depending on the physical configuration on the system, and these might be lost to the precision error if we are not rigorous with the tests.

Then if we are agreed, I will simply add the  `sparse=False` option to the tests involving eigenvalue solutions in the branch and close the issue.

Thanks for the quick responses, and all the best!
Vergílio

raphaeltimbo, 2021-07-08:
> Then if we are agreed, I will simply add the sparse=False option to the tests involving eigenvalue solutions in the branch and close the issue.

That is ok for me!

raphaeltimbo, 2021-08-24:
Closed with #826.

closed 2021-08-24 by raphaeltimbo