petrobras/ross #819: Error in cylindrical bearing when changing the number of volumes

raphaeltimbo, 2021-07-06:
I have tried the following code:

```python
    bearing = THDCylindrical(
        L=0.263144,
        R=0.2,
        c_r=1.95e-4,
        n_theta=11,
        n_z=3,
        n_y=None,
        n_gap=1,
        betha_s=176,
        mu_ref=0.02,
        speed=94.24777960769379,
        Wx=0,
        Wy=-112814.91,
        k_t=0.15327,
        Cp=1915.24,
        rho=854.952,
        T_reserv=50,
        fat_mixt=0.52,
        T_muI=50,
        T_muF=80,
        mu_I=0.02,
        mu_F=0.01,
        sommerfeld_type=2,
    )
    bearing.run([0.1, -0.1])
```

But I am getting this error:
```
    
>       auxFx = auxP * vector_auxF_x
E       ValueError: operands could not be broadcast together with shapes (3,22) (23,)

../fluid_flow/cylindrical.py:800: ValueError
```

Here I have decreased the number of volumes in the `test_cylindrical.py` files from:
```python
        n_theta=41,
        n_z=5,
```
To:
```python
        n_theta=11,
        n_z=3,
```

Any idea on what is the problem here?

@verg1lio , @elyqg

elyqg, 2021-07-13:
Hi @raphaeltimbo. I will check this problem and fix it. With the insertion of the pocket something was probably affected.

raphaeltimbo, 2021-07-15:
Thanks @elyqg ! I will assign the issue to you.

closed 2021-10-06 by raphaeltimbo