petrobras/ross #821: Mode shape tracking and identification for the Campbell diagram

verg1lio, 2021-07-07:
Dear all, good afternoon

This issue is a proposition for an enhancement of the Campbell diagram's capabilities, attending to what was discussed previously in a Petrobrás - EDGE meeting. The requirements asked are:

1 - ability to identify the mode shapes in an automated manner (_i.e._ is the mode a lateral one? axial? torsional?);
2 - ability to track the modes if they suffer crossing over or veering phenomena, without mixing them up.

The first is a "quality of life" improvement, and aids in the evaluation of mechanical systems. The second is needed especially in the case of 6 _DoF_ models, since lateral and torsional or lateral and axial modes tend to interact and present said phenomena.

After evaluation of current methodologies and viable manners to achieve these goals, I suggest we follow the proposed by the annexed article below. I will be busy in the near future developing this formulation and applying it to a known system, in order to verify its validity and code consistency. This will be done via a dedicated PR in our [https://github.com/LMEst-Rotor/ross](url) branch.

[ACriterionforModeShapeTracking-ApplicationtoCampbellDiagrams.pdf](https://github.com/ross-rotordynamics/ross/files/6779098/ACriterionforModeShapeTracking-ApplicationtoCampbellDiagrams.pdf)

verg1lio, 2021-07-07:
The pseudo-code is as follows:

[image]

raphaeltimbo, 2021-07-15:
Ok, @verg1lio ! I will be assigning the issue to you. 

PS.: don't worry about using 'WIP' in issues, we only use it in PRs to avoid merging work that it is not complete.

verg1lio, 2021-07-15:
> Ok, @verg1lio ! I will be assigning the issue to you.
> 
> PS.: don't worry about using 'WIP' in issues, we only use it in PRs to avoid merging work that it is not complete.

Thank you @raphaeltimbo , and I'll remember that too.

raphaeltimbo, 2024-06-20:
Identification of modes has been solved in #1066;

closed 2024-06-20 by raphaeltimbo