petrobras/ross #845: Static analysis quirks

LeifParr, 2021-11-15:
Hi all. I ran into a little quirk while doing an analysis of a horizontal axis multistage pump. I wanted to use the run_static method to produce reaction forces for bearing loads. A colleague of mine pointed out that the sum of reaction forces did not add up to the shaft weight + disk mass. It turns out that when I put a disk element in a node where I put a bearing-element, the disk element is not included in static analysis. 
Attached is my rotor as a TOML-file (had to name it .zip to upload)
[LeifParr_rotor_TOML.zip](https://github.com/ross-rotordynamics/ross/files/7541465/LeifParr_rotor_TOML.zip)

Here is a test script:
`
import ross as rs

# set plotly to render default in browser or spyder
import plotly.io as pio
pio.renderers.default='browser'  # 'browser' for browser, 'svg' for spyder

# load rotor from file
path = 'LeifParr_rotor_TOML.zip'
rotor = rs.Rotor.load(path)
# print total rotor mass and gravity force
print(f'Rotor mass (rotor.m): {rotor.m:.1f} kg, total rotor gravity force (rotor.m * 9.8065) {rotor.m*9.8065:.0f} N')
# plot rotor
rotor_plot_fig = rotor.plot_rotor(nodes=40)
rotor_plot_fig.update_layout(title=path)
#rotor_plot_fig.show()
# run static analysis
static_results = rotor.run_static()
print('static results, bearing forces')
a = static_results.bearing_forces
total_force = 0.0
for el in a:
    total_force += a[el]
print(f'static results, sum of bearing forces {total_force:.1f} N')
print(f'Difference: {rotor.m*9.8065 - total_force:.1f} N')
print(static_results.bearing_forces)

# make list of bearing nodes
bearing_nodes = []
for elem in rotor.bearing_elements:
    bearing_nodes.append(elem.n)
# sum all disk element masses which are also bearing nodes
disk_element_mass_bearings = 0.0
for elem in rotor.disk_elements:
    if elem.n in bearing_nodes:
        disk_element_mass_bearings += elem.m
print(f'Sum of all masses of disk elements which are also bearing nodes: {disk_element_mass_bearings * 9.8065:.1f} N')

# Also, the quirk is also present in plotting, disk elements are plotted as negative bearing reaction forces:
static_results.plot_free_body_diagram().show()
`
Here is the output:
_runfile('C:/Leif/github/Rotordynamics/static_bug.py', wdir='C:/Leif/github/Rotordynamics')
Rotor mass (rotor.m): 175.1 kg, total rotor gravity force (rotor.m * 9.8065) 1717 N
static results, bearing forces
static results, sum of bearing forces 1627.1 N
Difference: 90.2 N
{'node_17': 260.1, 'node_51': 116.9, 'node_88': 539.0, 'node_138': 711.1}
Sum of all masses of disk elements which are also bearing nodes: 90.3 N_

Here are rotor plot and free body diagram.
[image]
[image]

I think the bug lies in the run_static method in rotor_assembly.py, but I haven't found a quick way to fix it yet. It seems that in collecting bearing and disk forces, it is either a bearing force, or a disk force, can't be both.:
`
  # locate and collect bearing and disk forces
  aux_df = aux_rotor.df.loc[
      (aux_rotor.df["type"] != "ShaftElement")
      & (aux_rotor.df["shaft_number"] == i)
  ]
  for j, row in aux_df.iterrows():
      if row["n"] == n_max:
          force = -np.round(elm_forces_y[-1, 1], 1)
      else:
          force = np.round(elm_forces_y[int(row["n"]) - n_min, 0], 1)
  
      if row["type"] == "DiskElement":
          DskForce_nodal["node_" + str(int(row["n"]))] = force
          DskForce_tag[row["tag"]] = force
      elif row["type"] == "BearingElement":
          BrgForce_nodal["node_" + str(int(row["n"]))] = force
          BrgForce_tag[row["tag"]] = force
          if not pd.isna(row["n_link"]):
              BrgForce_nodal["node_" + str(int(row["n_link"]))] = -force
`

Suggestions for a quick (and lasting) fix?

raphaeltimbo, 2021-11-16:
Hi @LeifParr 
Thank you for the detailed report. 
I will try to look at it as soon as possible and will let you know if I find a solution.

LeifParr, 2021-11-16:
I was able to come up with a temporary work-around. It took some time for me to wrap my head around the code, especially the concept of having several shafts. Anyway, here is a temporary work-around (only for free-body-diagram) for a one-shaft situation:
```
# -*- coding: utf-8 -*-
"""
Created on Tue Nov 16 07:08:54 2021
Test script for static analysis
@author: lpa
"""

import pandas as pd 
import ross as rs
from collections import Counter
import numpy as np
from scipy import linalg as la

# load rotor from file
path = 'c:/data/rotordyn/LeifParr_rotor_TOML.zip'
rotor = rs.Rotor.load(path)
# make a new dummy rotor:
aux_brg = []
for elm in rotor.bearing_elements:
    aux_brg.append(rs.BearingElement(n=elm.n, kxx=1e14, cxx=0))
aux_rotor = rs.Rotor(rotor.shaft_elements, disk_elements=rotor.disk_elements, bearing_elements=aux_brg)
# print total rotor mass and gravity force
print(f'Rotor mass (aux_rotor.m): {aux_rotor.m:.1f} kg, total rotor gravity force (aux_rotor.m * 9.8065) {aux_rotor.m*9.8065:.0f} N')
#

# plot rotor
rotor_plot_fig = aux_rotor.plot_rotor(nodes=40)
rotor_plot_fig.update_layout(title=path)
# rotor_plot_fig.show()

# copy static analysis method from rotor.assembly.py:
aux_K = aux_rotor.K(0)

# gravity aceleration vector
g = -9.8065
grav = np.zeros(len(aux_rotor.K(0)))
grav[1 :: aux_rotor.number_dof] = g

# calculates u, for [K]*(u) = (F)
disp = (la.solve(aux_K, aux_rotor.M() @ grav)).flatten()
gravity_force = aux_rotor.M() @ grav

# calculate forces
nodal_forces = rotor.K(0) @ disp

reaction_forces = nodal_forces - gravity_force

reaction_forces_y = reaction_forces[1 :: aux_rotor.number_dof]
# print(reaction_forces_y)

# make dict of bearing forces
bearing_force_nodal = {}
for bearing in aux_rotor.bearing_elements:
    node = bearing.n
    force = reaction_forces_y[node]
    bearing_force_nodal[f'node_{node:d}'] = force
print(bearing_force_nodal)

# make a dict of disk forces
disk_force_nodal = {}
for disk in aux_rotor.disk_elements:
    node = disk.n
    force = disk.m*g*-1.0
    disk_force_nodal[f'node_{node:d}'] = force
print(disk_force_nodal)

# calculate total shaft weight for plot, assume only one shaft
shaft_w = 0.0
for shaft in aux_rotor.shaft_elements:
    shaft_w += shaft.m * g * -1.0
w_shaft = [shaft_w,]

# make nodes and nodes_pos arrays
nodes =[aux_rotor.nodes,]
nodes_pos = [aux_rotor.nodes_pos]

# initalize a dummy set of static results for plotting:
static_results = rs.StaticResults(deformation=None, Vx=None, Bm=None, w_shaft=w_shaft, 
                 disk_forces=disk_force_nodal, 
                 bearing_forces= bearing_force_nodal, nodes=nodes,
                 nodes_pos=nodes_pos, Vx_axis=None)

fig = static_results.plot_free_body_diagram()
fig.update_layout(title = 'Grane WIP. Static bearing load. 2021-11-16')
fig.show()
```

raphaeltimbo, 2021-11-17:
Great to hear that @LeifParr. 
As you noticed, the if block to check for `DiskElement` or `BearingElement` breaks the code.
I will try to use your code as a base to fix this.

Regarding your comment:
>  It took some time for me to wrap my head around the code, **especially the concept of having several shafts**. 

We were trying to run an analysis on coaxial rotors and we tried to change some parts of the code to deal with that, which definitely complicates the code a lot. 
My opinion today is that we should have dealt with the coaxial case in a separate function, and even though we would have some code repetition there, we would be able to keep the main code more simple. Maybe in the future I can find some time to refactor that as well.

raphaeltimbo, 2021-12-10:
Hi @LeifParr. I have merged a PR with some code based on your approach. 
I have also released the v1.1.0 version with the updated code. 
I will close the issue for now. Let me know if you have any problems.

closed 2021-12-10 by raphaeltimbo