petrobras/ross #876: bug with modelling worn bearing, when i create an object using the saved class -example 4 for worn bearing , there is no change in the eccentricity or the attitude angle with the change in wear depth.

gangadHARD1, 2022-02-21:


gangadHARD1, 2022-02-21:
from ross.fluid_flow.fluid_flow import fluid_flow_example4
wornbear=fluid_flow_example4()     #creating the object
wornbear.max_depth=30e-9         #changing the max depth
wornbear.displacement=.08725     #changing the angular displacement of the wear 
no matter  what value I set to parameters (max_depth and displacement) The value of the pressure matrix ,eccentircity  or the attitude angle doesnt change, ps help

gangadHARD1, 2022-02-21:
also with the change in the value of the parameter of the object from worn bearing class only changes the pressure matrix ,no change in the eccentricity with omega.

raphaeltimbo, 2022-03-17:
Hi @gangadHARD1.
The fluid_flow class does all the calculation when the bearing is instantiated, this means that latter changes to the bearing attributes will not trigger the calculation.

So, to get the results that you are looking for, you have to create a new bearing with the desired values:

```python
import ross as rs

nz = 8
ntheta = 32
omega = 1000 * np.pi / 30.0
p_in = 0.0
p_out = 0.0
radius_stator = (30 / 2) * 10 ** (-3)
cr = 90 * 10 ** (-6)
radius_rotor = radius_stator - cr
length = 20 * 10 ** (-3)
load = 18.9
viscosity = 1.044 / 10.0
density = 881
max_depth = 50 * 10 ** (-6)
y = 10 * np.pi / 180

worn_bearing = rs.fluid_flow.fluid_flow.FluidFlow(
    nz,
    ntheta,
    length,
    omega,
    p_in,
    p_out,
    radius_rotor,
    radius_stator,
    viscosity,
    density,
    load=load,
    shape_geometry="wear",
    max_depth=max_depth,
    displacement=y,
)

max_depth2 = 60 * 10 ** (-6)
worn_bearing2 = rs.fluid_flow.fluid_flow.FluidFlow(
    nz,
    ntheta,
    length,
    omega,
    p_in,
    p_out,
    radius_rotor,
    radius_stator,
    viscosity,
    density,
    load=load,
    shape_geometry="wear",
    max_depth=max_depth2,  # change max_depth when creating the bearing
    displacement=y,
)

print(worn_bearing.attitude_angle)
-> 0.38515248737114866
print(worn_bearing2.attitude_angle)
-> 0.3576283441280004
```

Fill free to reopen the issue if necessary.

closed 2022-03-17 by raphaeltimbo

gangadHARD1, 2022-03-18:
Thanks a ton!!!!!