petrobras/ross #885: Comparison of the natural frequency result of ROSS and ANSYS

WaCkySongs, 2022-04-23:
Hi, everyone,
  I'm studying the rotor dynamic with ANSYS and ROSS. I recently made a very simple benchmark test to compare the result of natural frequency of a shaft-beam (without any constrain or load). 
  However, the result shows big difference between ROSS (with 4DOF or 6DOF Timoshenko Beam) and ANSYS ( by default with BEAM189 ( 6DOF 3-node beam quadratic element ) or SOLID).
  Could you please give me some information on it, i. e., why the natural frequency obtained by ANSYS(w7~=1600) is much smaller than those of ROSS (w7~= 9500 for both BEAM186 or SOLID) ? Is there any correction for the BEAM 188 in ANSYS?

  Here is the setting for the all tests:
            1. Material: steal by default.
            2. length/inner/outer diammeter of the shaft : 30.08/7/300 [mm] respectively
            3. shaft is discritized into 10 elements. 
            4. no constrain is applied
            
  Here is the result 
[image]

  Here is the code in ROSS  

```python
steel = rs.Material(name="Steel", rho=7850, E=210e9, Poisson=0.3)

nShaft = 10
nL = 0.3 / nShaft
d_in = 7e-3
d_out = 30.08e-3

shaft = [
    rs.ShaftElement6DoF(
    L=nL,
    idl=d_in,
    odl=d_out,
    material=steel,
    shear_effects=True,
    rotary_inertia=True,
    gyroscopic=False,
    )
    for _ in range(nShaft)
]
# print(shaft)

#% other
r1 = rs.Rotor(shaft)
modal = r1.run_modal(0,24)
print('natural frequencies',modal.wn)
print('damped natural frequencies',modal.wd)
```

rodrigomoliveira1, 2022-04-23:
Hey @WaCkySongs! Thank you for doing such comparisons.

I believe both results are correct. The issue here is ANSYS returns the results in **HERTZ** unit, while ROSS returns the values in **RAD/S** as default.

So, if you converts ANSYS results to **rad/s**, you'll have `1536Hz * 6.28 (rad/s / Hz) = 9646 rad/s`, which is close enough to ROSS results.

Hope this helps!

WaCkySongs, 2022-04-23:
Hi, @rodrigomoliveira1. 
Thanks so much for your reply. It was my mistake that I did not get the right Unit.

closed 2022-04-23 by WaCkySongs