petrobras/ross #898: Parameter identification module

geovanebritto, 2022-08-24:
Dear @raphaeltimbo, my name is Geovane Brito, mechanical engineer student at UFBA and I’m being tutored for my graduation’s degree by professor @verg1lio, who you already know. I decided to focus my TCC on rotodynamics because I already work with rotating machinery vibration analysis, and that is a very interesting area for me. Since I knew Vergílio, he presented me ROSS and I promptly became interested in knowing more about the software. I’ve made some simulations on it and I liked it a lot. My practical experientes with on field applications made me ask if a few more resources wouldn’t make it even better.
This text is to present myself and ask you if I could contribute to ROSS with the support of Vergílio. Our ideia is to make a module for parameter identification in order to identify field case variables. It would use field data as inputs and compare it with a simulated machine behavior (to aid in retrofit, revamp or reevaluation applications, for example). Also to extract useful data to estimate fatigue life of shafts and structures, of possible. It would be an independent module, coded in the same fashion as the rest of the code made by LMEst researchers, and its development is intend to be started as of now and finished by mid October/22. We are of course open for conversations and adaptations of the proposed code and any other topic regarding it. It would be a pleasure to be able to help improve ROSS.

thiagoritto, 2022-08-25:
Dear @geovanebritto , we are building a team (UFU and UFRJ) to add new modules to ROSS. One of them is related to calibration (identification). Your initiative comes in good timing. You are more then welcome to proceed with your idea. We can also arrange a small online meeting to discuss it (my email is tritto@mecanica.ufrj.br). Thiago Ritto.

verg1lio, 2022-08-26:
Dear @thiagoritto , good morning! I was part of the UFU team, responsible for the 6 DoF model and some of the bearings added to ROSS. Since July/2022 I was approved as a Professor as UFBA, and @geovanebritto is my first TCC student. To me, ROSS is exceptionally applicable software, with which I have experience, and it felt natural to suggest that further research be made on it. 

I believe a chat would prove useful for everyone involved, as we do not want to overlay our module proposal with what is already being implemented by UFRJ or UFU.

To the best of my memory, this parameter identification module was not yet proposed, and both @geovanebritto and myself (@verg1lio) believe that it could prove to be a useful tool for rotordynamics analysis. It would automate something that ROSS already allows for, but demands a somewhat complicated manual coding process.

thiagoritto, 2022-08-29:
@verg1lio, I agree. Let's discuss it. My email is tritto@mecanica.ufrj.br.

closed 2022-08-29 by thiagoritto

raphaeltimbo, 2022-08-29:
This is great @geovanebritto and @verg1lio!
As pointed out by @thiagoritto, let's try to arrange a meeting and discuss things in more detail.