petrobras/ross #903: Of course, the example below shows an example of solving a transient forced response for the `.rotor_example()` Rotor object using Runge-Kutta 4th order. The `x` returned variable is the state space presentation of the system varying in time.

MAATALLAHDJEMAI, 2022-09-24:
Of course, the example below shows an example of solving a transient forced response for the `.rotor_example()` Rotor object using Runge-Kutta 4th order. The `x` returned variable is the state space presentation of the system varying in time.

I remind you as well that Ross has thes the `.solve_transient()` native method that can also be used. But in this case you would not explicitly use the state space formulation.

```
import ross as rs
import numpy.linalg as la
import numpy as np

def RK4(A, x0, t, Minv, f_mag, freq, dof_forced):
    
    x = np.zeros((len(A), len(t)))
    x[:,0] = x0[:]
    
    for i, ti in enumerate(t[:-1]):
        
        F = np.zeros((len(A), 3))
        
        dt = t[i+1] - ti
        
        F[len(A)//2+dof_forced, :] = np.array([f_mag * np.sin(2*np.pi * freq * ti),
                                            f_mag * np.sin(2*np.pi * freq * (t[i+1]+ti)/2),
                                            f_mag * np.sin(2*np.pi * freq * t[i+1])])
                                                        
        aux1 = np.zeros((len(A), 1))
        aux1[:, 0] = Minv @ F[:, 0]
        aux2 = np.zeros((len(A), 1))
        aux2[:, 0] = Minv @ F[:, 1]
        aux3 = np.zeros((len(A), 1))
        aux3[:, 0] = Minv @ F[:, 2]
    
        k1 = A @ x[:,i].reshape((x.shape[0], 1)) + aux1
        x1 = x[:,i].reshape((x.shape[0], 1)) + k1 * dt / 2
        k2 = A @ x1 + aux2 
        x2 = x[:,i].reshape((x.shape[0], 1)) + k2 * dt / 2
        k3 = A @ x2 + aux2
        x3 = x[:,i].reshape((x.shape[0], 1)) + k3 * dt
        k4 = A @ x3 + aux3
        
        x[:, i+1] = x[:,i] + dt / 6 * (k1 + 2 * k2 + 2 * k3 + k4)[:, 0]

    return x

rotor = rs.rotor_example()

sp = 10 # rotation speed
freq = 100 # excitation frequency

A = rotor.A(sp)
M = rotor.M()

Z = np.zeros(M.shape)
I = np.eye(len(M))

Minv = np.vstack([np.hstack([I, Z]),
                  np.hstack([Z, la.inv(M)])])

dt = 5e-5
tf = 5
t = np.arange(0, tf, dt)

dof_forced = 0 #degree of freedom to apply the force
force_mag = 10 # force magnitude in Newtons
x0 = np.zeros(len(A)) # Initial conditions

x = RK4(A=A, x0=x0, t=t, Minv=Minv, f_mag=force_mag,
        freq=freq, dof_forced=dof_forced)
```

_Originally posted by @atbrandao in https://github.com/ross-rotordynamics/ross/issues/841#issuecomment-1035322409_

MAATALLAHDJEMAI, 2022-09-24:
hi@atbrandao 
I have a problem, when I want to plot the frequency response, I encounter a mismatch error(ValueError: x and y must have same first dimension, but have shapes (1100,) and (56,)) please help me

raphaeltimbo, 2022-10-11:
Hi @MAATALLAHDJEMAI.
Take a look at the shape of the response. I believe it is (56, 100000) because calculation was carried out with the state-space matrix (A), so from 28 dof you have 28 displacements + 28 velocities.

So to plot results for the first node x coordinate using plotly for example you would need to do something like:

```python
import plotly.graph_objects as go

fig = go.Figure()
fig.add_scatter(
    x=t,
    y=x[0, :]
)
```

closed 2022-10-18 by raphaeltimbo