petrobras/ross #906: Magnitude plot unit

Rodr-Corre4, 2022-10-11:
Dear all, good afternoon,

I would like to ask about the output unit displayed in the magnitude plot while using the "rotor.run_freq_response" module.
The displacement output is really supposed to be returned in "m/N"?
If the answer is yes, would you kindly explain to me the physical meaning of this unit and its origin, please?

Best regards, 

[image]

raphaeltimbo, 2022-10-11:
Hi @Rodr-Corre4!
Taking the equation 2.79 from Friswell:

$$
[-\omega \mathbf{M} + j\omega \mathbf{C} + \mathbf{K}]\mathbf{q}_0 = \mathbf{Q}_0
$$

$$
\mathbf{q}_0 = [-\omega \mathbf{M} + j\omega \mathbf{C} + \mathbf{K}]^{-1}\mathbf{Q}_0
$$

Where $\mathbf{q}(t)$ is the generalized displacement and $\mathbf{Q} (t)$ the generalized force.
The matrix $\mathbf{D}(\omega) = [-\omega \mathbf{M} + j\omega \mathbf{C} + \mathbf{K}]^{-1}$ is the receptance matrix, which is what we are actually calculating with the `.run_freq_response` method. Since this result is not yet multiplied by the generalized force $\mathbf{Q}_0$ the units will be `m/N`.

So, the main problem with this plot is actually the naming of the y axis: Displacement. This name would only make sense with units like `m`.
I think we should replace this with something like Magnitude.

marcusfilipesr, 2023-03-13:
Hi @raphaeltimbo, was trying to solve this issue, and noticed that are other types of units, such as:

- Displacement (m/N)
- Velocity (m/s/N)
- Acceleration (m/s**2/N)

What do you think of changing all this to:

- Magnitude (m/N)
- Magnitude (m/s/N)
- Magnitude (m/s**2/N)

raphaeltimbo, 2023-03-14:
> Hi @raphaeltimbo, was trying to solve this issue, and noticed that are other types of units, such as:
>
>     * Displacement (m/N)
>
>     * Velocity (m/s/N)
>
>     * Acceleration (m/s**2/N)
>
>
> What do you [...]

Hi @marcusfilipesr. I agree with your solution. Lets proceed that way.

PR #954 (2023-03-14): Freq Response Magnitude label changed
This PR resolves #906
closed 2023-03-14 by raphaeltimbo