petrobras/ross #936: Add mass terms to bearing and seal elements

raphaeltimbo, 2023-01-02:
In pump seals mass terms can have a significant effect, especially for long seals where the direct mass coefficient can be quite large.
We should have the option to add the mass terms in the generic 
bearing and seal classes (`BearingElement` and `SealElement`).

marcusfilipesr, 2023-03-14:
I'm checking how the implementation will proceed. As soon as we found the path, I'll put it here.

raphaeltimbo, 2023-03-14:
Ok @marcusfilipesr.
I think we should basically follow the same logic used for stiffness and damping coefficients, basically replicating the code for mxx, mxy, myx and myy.

marcusfilipesr, 2023-03-14:
> Ok @marcusfilipesr. I think we should basically follow the same logic used for stiffness and damping coefficients, basically replicating the code for mxx, mxy, myx and myy.

We are thinking of something similar to it. Also, I was wondering if it is necessary a case in which I can built a test for the cases where the masses are passed, what do you think?

raphaeltimbo, 2023-03-14:
I think we should have at least some basic tests like:
- Check if the BearingElement.M() method returns the correct matrix;
- Check if the Rotor.M() method returns the correct matrix for a rotor with a bearing element with mxx, myy != 0.

marcusfilipesr, 2023-03-14:
> I think we should have at least some basic tests like:
> 
> * Check if the `BearingElement.M()` method returns the correct matrix;
> * Check if the `Rotor.M()` method returns the correct matrix for a rotor with a bearing element with `mxx, myy != 0`.

Alright!! I'm going to implement that right away!