petrobras/ross #955: Fix units for material density

wigging, 2023-03-15:
In the docstring for `ross.Material` the density `rho` is defined with units of `N/m**3` but in `Material.__str__` the units are `kg/m**3`. Based on the examples I've run, I believe the units for `rho` are `kg/m**3`. Also, the units should be written as `kg/m^3` instead of `kg/m**3`.

marcusfilipesr, 2023-03-15:
Hi @wigging, I believe you're right about the units. I'll be correcting this in PR that I'm working on. Regarding the power notation of the units, what do you think @raphaeltimbo?

wigging, 2023-03-15:
Python code uses double `**` to raise things to a power such as `2**5`. But for comments in the code I would recommend `2^5`.

raphaeltimbo, 2023-03-15:
Thanks for the issue @wigging!

You are correct, the unit should be `kg/m**3`. 

Regarding the discussion between `kg/m**3` and `kg/m^3`, the units library that we use (pint) accepts both, and it also accepts `kg/m³` (alt + 3 => ³). 

But I tend to prefer `kg/m**3` since this is more python like and one can create the quantity with pint with both of these options:

```python
Q_(1, 'kg / m ** 3')
# or
1 * ureg.kilogram / ureg.meter**3
```

This would be for the `Parameters` part of the documentation.
@marcusfilipesr, if we would add some latex to the documentation, I think it should be only 
in the initial part of the docstring (before `Parameters`). In that case we would have to use '^' (see the `ShaftElement` as example).

Also, I think we could improve the Material docstring with:

- Add 'optional' to some arguments;
- Add example using pint Q_.

marcusfilipesr, 2023-03-16:
> 

Okay! I'll open de PR with the corrections.

PR #958 (2023-03-17): Material density unit
This PR resolves #955
closed 2023-03-17 by raphaeltimbo