petrobras/ross #962: Can't access variables for plotting mode shapes

wigging, 2023-03-21:
The example below uses Matplotlib to plot the critical speeds using `wn` from the results which is a `UCSResults` object. I would also like to use Matplotlib to plot the mode shapes from the `ModalResults` object but I don't see any variables that are available to plot the relative displacement of the mode shapes. I know there are plot functions in ROSS that use Plotly, but Matplotlib is more useful for generating custom plots and for generating plots for publications.

```python
import matplotlib.pyplot as plt
import numpy as np
import ross as rs

# Steel material properties
steel = rs.Material(name='Steel', rho=7810, E=211e9, G_s=81.2e9)

# Number of shaft elements
N_elem = 6

# Dimensions of a shaft element
L_total = 2.032
L = L_total / N_elem
i_d = 0
o_d = 0.127

shaft_elem = rs.ShaftElement(
    L=L,
    idl=i_d,
    odl=o_d,
    material=steel,
    shear_effects=True,
    rotary_inertia=True,
    gyroscopic=True
)

shaft_elements = [shaft_elem for _ in range(N_elem)]

# Bearings at beginning and end of shaft
bearing0 = rs.BearingElement(n=0, kxx=1.75126e8, cxx=0)
bearing1 = rs.BearingElement(n=N_elem, kxx=1.75126e8, cxx=0)
bearings = [bearing0, bearing1]

# Rotor assembly
rotor = rs.Rotor(shaft_elements=shaft_elements, bearing_elements=bearings)

# Results
stiff_range = (5, 9)
freq_range = (10.47, 3141.59)
results = rotor.run_ucs(stiffness_range=stiff_range, num_modes=12)

# Plot results using Matplotlib
wn0 = results.wn[0]
wn1 = results.wn[1]
wn2 = results.wn[2]
kb = results.stiffness_log

_, ax = plt.subplots(tight_layout=True)
ax.loglog(kb, wn0, label='1st')
ax.loglog(kb, wn1, label='2nd')
ax.loglog(kb, wn2, label='3rd')
ax.set_xlabel('Bearing stiffness [N/m]')
ax.set_ylabel('Critical speed [rad/s]')
ax.legend()

plt.show()
```

raphaeltimbo, 2023-03-21:
Hi @wigging !
For now, the data used to create each plotly figure for the mode shapes is available inside the `Shape` objects. Currently, these `Shape` objects are not well documented in our website, but you can take a look at the [plot_2d](https://github.com/petrobras/ross/blob/ac9f339e5939365f6907b1abc9fc567f35247e0d/ross/results.py#L518) and [plot_3d ](https://github.com/petrobras/ross/blob/ac9f339e5939365f6907b1abc9fc567f35247e0d/ross/results.py#L592) methods to see how the data is accessed to create the plots.

Another solution would be to access the data using the plotly figure. Something like:

```
modal = rotor.run_modal(speed=0)
fig = modal.plot_mode_2d(mode=0)

# get x and y to use in matplotlib
x = fig.data[0]['x']
y = fig.data[0]['y']
```

The `fig.data` object has this information in it:
```
(Scatter({
     'customdata': array([2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182,
                          2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182,
                          2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182,
                          2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182,
                          2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182, 2.75254182]),
     'hovertemplate': 'Displacement: %{y:.2f}<br>Angle rad: %{customdata:.2f}',
     'line': {'color': '#1f77b4'},
     'mode': 'lines',
     'name': 'major',
     'showlegend': False,
     'x': array([0.        , 0.08466667, 0.16933333, 0.254     , 0.33866667, 0.33866667,
                 0.42333333, 0.508     , 0.59266667, 0.67733333, 0.67733333, 0.762     ,
                 0.84666667, 0.93133333, 1.016     , 1.016     , 1.10066667, 1.18533333,
                 1.27      , 1.35466667, 1.35466667, 1.43933333, 1.524     , 1.60866667,
                 1.69333333, 1.69333333, 1.778     , 1.86266667, 1.94733333, 2.032     ]),
     'y': array([0.1426755 , 0.46738637, 0.78815069, 1.09797156, 1.3898521 , 1.3898521 ,
                 1.65978968, 1.9050767 , 2.12065876, 2.30148145, 2.30148145, 2.44455308,
                 2.54902502, 2.61305741, 2.63481039, 2.63481039, 2.61305741, 2.54902502,
                 2.44455308, 2.30148145, 2.30148145, 2.12065876, 1.9050767 , 1.65978968,
                 1.3898521 , 1.3898521 , 1.09797156, 0.78815069, 0.46738637, 0.1426755 ])
 }),
 Scatter({
     'hoverinfo': 'none',
     'line': {'color': 'black', 'dash': 'dashdot'},
     'mode': 'lines',
     'name': 'centerline',
     'showlegend': False,
     'x': array([0.        , 0.33866667, 0.67733333, 1.016     , 1.35466667, 1.69333333,
                 2.032     ]),
     'y': array([0., 0., 0., 0., 0., 0., 0.])
 }))
```

I think that with this data you would be able to replicate the plots with matplotlib.

wigging, 2023-03-21:
It looks like `fig.data` provides the major axis data. Is that correct? I don't want the major axis data points which are always positive, I want the data associated with the x and y axis as discussed in #960.

wigging, 2023-03-21:
If it helps, I can draft a pull request with some changes to the `ModalResults` class.

raphaeltimbo, 2023-03-22:
> If it helps, I can draft a pull request with some changes to the `ModalResults` class.

That would be great @wigging !

Regarding the solution for this, I think we have two options:

- Adding attributes as you have mentioned;
- Creating intermediate 'data' methods.

We have discussed this second one in the past and we have implemented it for some methods, but we were not able to make that for all the plot methods.

Basically, you would create a `data` method with the same signature as the plot method. This method would return a dataframe with all the data that is used to create the plots.

Here is some discussion about this:
https://github.com/petrobras/ross/issues/784#issuecomment-797690867

And here you can see an example of implementation:
https://github.com/petrobras/ross/pull/785/files

I am ok with both solutions.

marcusfilipesr, 2023-05-04:
Hi everyone, I've just assigned @ViniciusTxc3 to this issue, and he'll proceed with the PR to implement the intermediate method for getting the data.

PR #998 (2023-08-11): Add data_mode()
This PR resolves #962  . Created  ` def data_mode ` in `Model_Results` class for access via df.
closed 2023-08-11 by raphaeltimbo