petrobras/ross #964: Change how we normalize the vectors in Shape objects

raphaeltimbo, 2023-03-28:
`Shape` objects have the option to normalize the vector which is passed in instantiation. This is used when creating shape objects for the mode shapes.
It would make sense that the max major axis in each normalized mode shape would be equal to one, but that is not the case for our current code.

marcusfilipesr, 2023-04-20:
@raphaeltimbo, do you mean that instead of normalizing by the maximum number (either x or y), this process should occur for each direction, i.e., `x dofs` normalized by `max(x)` and same on to `y dofs`?

raphaeltimbo, 2023-04-24:
Hi @marcusfilipesr !
Basically, the plot_2d_mode, when called with orientation="major", should return a plot with the maximum value equal to 1. The way we have normalization now this does not happen.

marcusfilipesr, 2023-04-24:
I'm trying to reproduce, but I'm getting this behavior when choosing the `'y'` orientation.

marcusfilipesr, 2023-04-24:
I was looking at the code, and I the normalization is done here:

```python
if self.normalize:
    modex = evec[0::4]
    modey = evec[1::4]
    xmax, ixmax = max(abs(modex)), np.argmax(abs(modex))
    ymax, iymax = max(abs(modey)), np.argmax(abs(modey))

    if ymax > 0.4 * xmax:
        evec /= modey[iymax]
    else:
        evec /= modex[ixmax]
```

I am trying to understand the reason behind this `0.4` value, because when it is removed the results became:

[image]

Is it possible to use this solution?

raphaeltimbo, 2023-04-28:
I think in this case the major axis is aligned with x or y, so it works...
But I if the major axis is not aligned I think we could have problems. 
I can't think of how to build an example of that case using modal analysis (we will have that in forced cases), so maybe, for now we should build a 'artificial' eigenvector where the major axis is not aligned with the x and y directions to test how we will normalize things.

raphaeltimbo, 2023-05-02:
Actually, I was thinking that your suggestion of just removing the 0.4 would be the better option.
The modal analysis is the only case where we really want to normalize the vector (forced analysis we don't want that), and in the modal case I think we always will have the major axis aligned with x or y, so I believe we can go with this solution.

PR #972 (2023-05-11): Shape Normalization
This PR resolves #964
closed 2023-05-11 by raphaeltimbo