petrobras/ross #968: Slowest tests

raphaeltimbo, 2023-04-27:
These are our 10 slowest tests:

```
============================== slowest 10 durations ===============================
38.86s call     ross/rotor_assembly.py::ross.rotor_assembly.Rotor.run_critical_speed
20.82s call     ross/tests/test_rotor_assembly.py::test_ucs_rotor9
16.53s call     ross/stochastic st_rotor_assembly.py::ross.stochastic.st_rotor_assembly.ST_Rotor.run_campbell
15.02s call     ross/rotor_assembly.py::ross.rotor_assembly.rotor_example_6dof
10.68s call     ross/tests/test_results.py::test_save_load_freqresponse
10.38s call     ross/rotor_assembly.py::ross.rotor_assembly.Rotor.run_campbell
7.74s call     ross/rotor_assembly.py::ross.rotor_assembly.Rotor.run_freq_response
7.29s call     ross/stochastic/st_results.py::ross.stochastic.st_results.ST_Results.load
7.26s call     ross/stochastic/st_results.py::ross.stochastic.st_results.ST_Results.save
6.98s call     ross/tests/test_rotor_assembly.py::test_run_critical_speed
```

Some are actually doctests. I think we could work to decrease at least 1 minute here.

ViniciusTxc3, 2023-10-23:
I'm working on test with the longest time (_ross > rotor_assembly.py > run_critical_speed_), I noticed that the test takes time due to the calculation in "speed_range`.

[image]

I did some tests changing the `num_modes `but we didn't see a significant reduction (loss of _ms_).

[image]

@raphaeltimbo, would you prefer not to modify this test or remove from the docstring?

raphaeltimbo, 2023-10-25:
@ViniciusTxc3 , maybe we could change to something like `speed_range=(100, 300)` and see if we get an improvement?

ViniciusTxc3, 2023-10-25:
@raphaeltimbo I compared the times between the two approaches, but we have practically the same result:

[image]

This is a costly method in processing time, do you think it should remain?

ViniciusTxc3, 2023-10-27:
@raphaeltimbo, I'm making these changes to the functions to reduce testing time, in PR #1019 :

- call  _ross/tests/test_rotor_assembly.py::test_ucs_rotor9_:
`ucs_results = rotor9.run_ucs(num_modes=32`) to `ucs_results = rotor9.run_ucs(num_modes=8)`
the time: 20.82s to 9.24s

- call     _ross/rotor_assembly.py::ross.rotor_assembly.rotor_example_6dof_:
`camp6 = rotor6.run_campbell(np.linspace(0, 400, 101), frequencies=18)` to
`camp6 = rotor6.run_campbell(np.linspace(0, 400, 101), frequencies=6)`
the time: 15.02s to 4.96s

- call     _ross/tests/test_results.py::test_save_load_freqresponse_:
`speed = np.linspace(0, 1000, 51)` to `speed = np.linspace(0, 1000, 11)`
the time: 10.68s to 2.92s

- call     _ross/rotor_assembly.py::ross.rotor_assembly.Rotor.run_campbell_:
`speed = np.linspace(0, 400, 101)` to `speed = np.linspace(0, 400, 11)`
the time: 10.38s to 1.49s

---------------------------------------------
I couldn't identify this test:
16.53s call     ross/stochastic st_rotor_assembly.py::ross.stochastic.st_rotor_assembly.ST_Rotor.run_campbell

Now, I'm going to start looking at tests under 10s.

raphaeltimbo, 2023-11-16:
> @raphaeltimbo I compared the times between the two approaches, but we have practically the same result:
> 
> [image]
> 
> This is a costly method in processing time, do you think it should remain?

Hi @ViniciusTxc3. Sorry for the late response.
Let's remove it.

closed 2024-04-04 by ViniciusTxc3