petrobras/ross #985: Plots for magnitude are not smooth

raphaeltimbo, 2023-05-26:
As we can see in the figure:

[image]

Magnitude plots for forced response analysis are not smooth.
Maybe we could use some interpolation process to make it look better.

marcusfilipesr, 2023-06-07:
Hello, @raphaeltimbo!!

I was thinking in a way to solve this issue, and I found that the Plotly do have an interpolation between the points, using the `line_type="spline"` argument.

I'm adding this to a PR and, would like to ask if you can test it with this plot that you generated.

raphaeltimbo, 2023-07-07:
Hello @marcusfilipesr !
I link the option of using `line_shape="spline"`, but discussing with some colleagues, we were also thinking about modifying the way that we generate the data for the plot.
These colleagues suggested to implement some algorithm that would check where we have the natural frequencies, and in that region we would also have more 'samples' to calculate the amplitude, since a small number of samples in that region can mislead us in which is the maximum value that we can achieve (e.g. if one speed sample is exactly at the peak we get the maximum value).

marcusfilipesr, 2023-07-31:
Hi @raphaeltimbo!

Sorry for the delayed answer! I'll start idealizing an algorithm according to your suggestion.

marcusfilipesr, 2023-10-24:
@raphaeltimbo to create this new samples, with the code as is, it would be necessary to add more elements to the rotor, correct?

I think a possible solution would be a "copy rotor" inside the one you're running the frequency response analysis, where you could add more elements as you please.

Am I getting it right? Or I'm missunderstanding it?

marcusfilipesr, 2023-10-25:
> Hi @marcusfilipesr! For the 'samples' I was actually referring to the frequency array used in frequency/forced response. For example, in the plot here in this issue we have an analysis from 0 to 15000 RPM. This analysis is carried out with the user providing an array of frequencies that will be equally spaced between each frequency value (a `np.linspace` for example).
> 
> My idea is to get this array and in the region around the natural frequency (5k to 7k in our example) have more frequencies being evaluated.
> 
> If the array had a space of 5 RPM between each value, in this region we could have 1 RPM, or some value that could even change as we get closer to the peak.

Got it! Sorry for the lack of understanding. I'll do that.

raphaeltimbo, 2024-03-22:
We should test the implemented method cluster_points to evaluate if it can solve this issue.
If that is the case, I think we should change the documentation to make it more clear what this option does.

ViniciusTxc3, 2024-05-09:
I tested including the "cluster points" method but the effect was not as expected.
[image]
This was the test code:
```python
rotor = rs.rotor_example()
speed = np.linspace(0, 1000, 101)
response = rotor.run_freq_response(speed_range=speed, cluster_points=True, num_points=5)
fig = response.plot_magnitude(inp=13, out=13, frequency_units="RPM").show()
```
@raphaeltimbo, I believe we will need to refine this region to improve visualization.

jguarato, 2024-05-20:
> As we can see in the figure: [...]

@raphaeltimbo,
Do you still have the script to generate this graph so I can investigate this issue further?