petrobras/ross #999: Error regarding the transformation matrix used in converting the angle of the probes in ROSS plots and calculations.

stanley-washington, 2023-07-26:
When adjusting the angle of the probes in the ross, a transformation matrix called 'operator' is used. However, it has an error:

[image]

And the correct form should be:

[image]

ViniciusTxc3, 2023-10-04:
These matrices are also in the _abs_defect.py_ and _st_results.py_ files, in addition to _results.py_. I created PR #1002 to make this correction.

raphaeltimbo, 2024-03-14:
Hi @ViniciusTxc3 and @stanley-washington 
I think we should recheck this issue based on discussion #1033.
@stanley-washington, could you provide some reference for the modification proposed here?
Also, we should add some tests to verify the values that we get from run_time_response depending on probe orientation so that we have this part of the code covered.

raphaeltimbo, 2024-03-14:
Some tasks related to this issue:

- [ ] Extract the probe response calculation from the plot method and create a probe_response method;
- [ ] Probe response method should return a dataframe with time column and response column for each probe passed in the list;
- [ ] Add test for probe_response method.

```python
def probe_response(self, probe):
    ...
    return probe_response

def plot_1d(self, probe, ...):
    ...
    probe_response = self.probe_response(probe)
    ...
    fig.add_scatter(
        x=probe_response["t"]
        y=probe_response["probe_y"]
```

stanley-washington, 2024-03-19:
The operation described entails rotating the predicted displacement values based on the angle specified by the argument probe (_p[1]_). This rotation is achieved using a transformation matrix, denoted as _T_, where the _X_ and _Y_ coordinate axes are rotated accordingly. However, there's an issue with the implementation of the transformation matrix _T_ in the API, as evidenced in lines 4440-4444 of the '_results.py_' file.

[image]

[image]

Currently, the operation simultaneously calculates the rotated values for both the _X_ and _Y_ axes. However, for the purpose of visualizing the signal rotated with respect to the probe, it suffices to consider only the rotation of the _X_ axis.

Furthermore, a sum-of-squares operation on the _X_ and _Y_ values is being performed, which is unnecessary for the intended purpose. Consequently, this step can be omitted from the procedure.

In view of this, I believe that the correct form for the code would be as follows:

[image]

ViniciusTxc3, 2024-03-19:
@stanley-washington @raphaeltimboI created a PR for this fix.
I ran the example and the result was this:

```python
probe1 = (3, 0)   # node 3, orientation 0° (X dir.)
probe2 = (3, 90)  # node 3, orientation 90°(Y dir.)
probe3 = (3, 45)  # node 3, orientation 45°(Y dir.)

response3.plot_1d(probe=[probe1, probe2, probe3], probe_units="degree").show()
```

[image]

Now, I will work on the method to return the dataframe and tests.

ViniciusTxc3, 2024-03-20:
@raphaeltimbo I updated PR #1037, with the suggested items:

- [x] Extract the probe response calculation from the plot method and create a probe_response method;
- [x] Probe response method should return a dataframe with time column and response column for each probe passed in the list;
- [x] Add test for probe_response method.

The new intermediate method is called `data_probe_response()`.

PR #1002 (2024-01-24): Converting angle of the probe
This PR resolves #999
closed 2024-04-18 by ViniciusTxc3