petrobras/ross #1010: Add critical speed to campbell modal plot

raphaeltimbo, 2023-10-10:
In the campbell + modal plot, if we click a critical speed point in the map like the following:
[image]

We get this error:

```python
--------------------------------------------------------------------------
KeyError                                 Traceback (most recent call last)
File ~\Anaconda3\envs\rs\lib\site-packages\ipywidgets\widgets\widget.py:766, in Widget._handle_msg(self, msg)
    764         if 'buffer_paths' in data:
    765             _put_buffers(state, data['buffer_paths'], msg['buffers'])
--> 766         self.set_state(state)
    768 # Handle a state request.
    769 elif method == 'request_state':

File ~\Anaconda3\envs\rs\lib\site-packages\ipywidgets\widgets\widget.py:643, in Widget.set_state(self, sync_data)
    638         self._send(msg, buffers=echo_buffers)
    640 # The order of these context managers is important. Properties must
    641 # be locked when the hold_trait_notification context manager is
    642 # released and notifications are fired.
--> 643 with self._lock_property(**sync_data), self.hold_trait_notifications():
    644     for name in sync_data:
    645         if name in self.keys:

File ~\Anaconda3\envs\rs\lib\contextlib.py:142, in _GeneratorContextManager.__exit__(self, typ, value, traceback)
    140 if typ is None:
    141     try:
--> 142         next(self.gen)
    143     except StopIteration:
    144         return False

File ~\Anaconda3\envs\rs\lib\site-packages\traitlets\traitlets.py:1498, in HasTraits.hold_trait_notifications(self)
   1496 for changes in cache.values():
   1497     for change in changes:
-> 1498         self.notify_change(change)

File ~\Anaconda3\envs\rs\lib\site-packages\ipywidgets\widgets\widget.py:694, in Widget.notify_change(self, change)
    691     if name in self.keys and self._should_send_property(name, getattr(self, name)):
    692         # Send new state to front-end
    693         self.send_state(key=name)
--> 694 super().notify_change(change)

File ~\Anaconda3\envs\rs\lib\site-packages\traitlets\traitlets.py:1513, in HasTraits.notify_change(self, change)
   1511 def notify_change(self, change):
   1512     """Notify observers of a change event"""
-> 1513     return self._notify_observers(change)

File ~\Anaconda3\envs\rs\lib\site-packages\traitlets\traitlets.py:1560, in HasTraits._notify_observers(self, event)
   1557 elif isinstance(c, EventHandler) and c.name is not None:
   1558     c = getattr(self, c.name)
-> 1560 c(event)

File ~\Anaconda3\envs\rs\lib\site-packages\plotly\basewidget.py:722, in BaseFigureWidget._handler_js2py_pointsCallback(self, change)
    719 trace = self.data[trace_ind]
    721 if event_type == "plotly_click":
--> 722     trace._dispatch_on_click(points, state)
    723 elif event_type == "plotly_hover":
    724     trace._dispatch_on_hover(points, state)

File ~\Anaconda3\envs\rs\lib\site-packages\plotly\basedatatypes.py:6121, in BaseTraceType._dispatch_on_click(self, points, state)
   6117 """
   6118 Dispatch points and device state all all hover callbacks
   6119 """
   6120 for callback in self._click_callbacks:
-> 6121     callback(self, points, state)

File c:\users\up49\ross\ross\results.py:1633, in CampbellResults.plot_with_mode_shape.<locals>._plot_with_mode_shape_callback(trace, points, state)
   1630 natural_frequency = trace.y[point_idx][0]
   1632 # run modal analysis for desired frequency
-> 1633 modal = self.modal_results[Q_(frequency, frequency_units).to("rad/s").m]
   1635 # identify index of desired mode
   1636 idx = (
   1637     np.abs(
   1638         modal.wd - Q_(natural_frequency, frequency_units).to("rad/s").m
   1639     )
   1640 ).argmin()

KeyError: 703.7547974962505
```

jguarato, 2024-04-11:
Is this still a problem? I tried to reproduce, but didn't get this error.
[image]

raphaeltimbo, 2024-04-11:
Hi @jguarato!
We still have the error. Sorry, the initial message in the issue was not clear on how to reproduce it.
The error occurs when using the plot_with_mode_shape method. 
You can run the code below and try to click the critical speed to reproduce it.

```python
import ross as rs
import numpy as np

rotor = rs.rotor_example()
camp = rotor.run_campbell(speed_range=np.linspace(0, 1000))
camp.plot_with_mode_shape()

jguarato, 2024-04-12:
No worries, @raphaeltimbo.

Now, I was able to reproduce the error:

```python
File ~/Documents/GitHub/ROSS_Code/ross/ross/results.py:1634, in CampbellResults.plot_with_mode_shape.<locals>._plot_with_mode_shape_callback(trace, points, state)
   1631 natural_frequency = trace.y[point_idx][0]
   1633 # run modal analysis for desired frequency
-> 1634 modal = self.modal_results[Q_(frequency, frequency_units).to(\"rad/s\").m]
   1636 # identify index of desired mode
   1637 idx = (
   1638     np.abs(
   1639         modal.wd - Q_(natural_frequency, frequency_units).to(\"rad/s\").m
   1640     )
   1641 ).argmin()
```

Looking at the code I noticed that this error appears because in line 1634, `frequency` does not correspond to any key in `self.modal_results`. I believe we would need to call `run_modal()` function for each considered frequency. 

So, as a solution, I thought about passing this function as an argument in `CampbellResults`, but I don't know if that would be the best way.
[image]
[image]

By making this change, the error no longer appears:
[image]

What do you think?

PR #1050 (2024-04-25): Fix click callback function of campbell plot with mode shape
Closes #1010
closed 2024-04-25 by raphaeltimbo