petrobras/ross #1011: Plotly problems

EDrotor91, 2023-10-13:
Hey Everyone,

i'm quite a beginner in using python. 
I have installed anaconda including python and istalled all needed packages to run the ross programm.
I have launched the tutorial script below in spyder. No errors occured but i can't see the plot of the rotor.
I tried to change the renderers to "browser" , "svg" , "png".... but it is not showing a plot in spyder or in the browser.
Could someone please explain me how to setup spyder, or maybe should i use another IDE?

from pathlib import Path
import ross as rs
import numpy as np
import plotly.graph_objects as go
import plotly.io as pio
#import plotly.express as go

pio.renderers.default = "notebook"

shaft_file = Path("shaft_si.xls")
shaft = rs.ShaftElement.from_table(
    file=shaft_file, sheet_type="Model", sheet_name="Model"
)

file_path = Path("shaft_si.xls")
list_of_disks = rs.DiskElement.from_table(file=file_path, sheet_name="More")

bearing1 = rs.BearingElement.from_table(n=7, file="bearing_seal_si.xls")
bearing2 = rs.BearingElement.from_table(n=48, file="bearing_seal_si.xls")

bearings = [bearing1, bearing2]

rotor3 = rs.Rotor(shaft, list_of_disks, bearings)

node_increment = 5
rotor3.plot_rotor(nodes=node_increment)

static = rotor3.run_static()

Best Regards 
Eric

shishirsanjeeva, 2023-10-16:
Hi @EDrotor91
I am not sure on how this can be solved in Spyder IDE. But you can use Jupyter notebook where you don't get these errors.

raphaeltimbo, 2023-10-16:
Hi @EDrotor91. 
You can use this code:

```python
import plotly.io as pio
pio.renderers.default = "browser"

fig = rotor3.plot_rotor(nodes=node_increment)
fig.show()
```

EDrotor91, 2023-10-17:
The main problem i had was not to call up the figure again...
Now i use VSCode as IDE with the jupyter notebook extension.
Everything works fine now :)

closed 2023-12-21 by raphaeltimbo