petrobras/ross #1017: Crack_example does not work

romero-patrick, 2023-10-26:
Good morning!
I tried to use the crack_example function but the following error occurs

No such file or directory: '/usr/local/lib/python3.10/dist-packages/tools/data/PAPADOPOULOS.csv'

I used other functions from the defects class but this error only occurs when I use the crack function.

Thank you.

ViniciusTxc3, 2023-10-31:
@romero-patrick currently to run the _"defects"_ class (in the future it will be called _"malfunctions"_), it needs to be run on your machine with the ross clone. Because in Jupyter or Colab, these folders are not being assigned.

[image]

@raphaeltimbo might know another solution.

raphaeltimbo, 2023-10-31:
We can add the file to the package distribution. This way, after a pip install the function will run without errors.

ViniciusTxc3, 2024-03-26:
@raphaeltimbo @romero-patrick We have added the necessary files for this type of testing in PR #1038.

closed 2024-04-04 by ViniciusTxc3