petrobras/ross #1020: Evaluate use of pypardiso

raphaeltimbo, 2023-11-01:
Maybe we could use this package to improve our performance: https://pypi.org/project/pypardiso/

jguarato, 2024-01-31:
Hi @raphaeltimbo, I took a look at the PyPardiso package.

While researching, I found that its solver is faster than SciPy solver, especially for very large systems [[Reference]](https://newtonexcelbach.com/2021/10/17/making-finite-element-analysis-go-faster-update-and-pypardiso/).
So, I decided to run some tests on ROSS to analyze the time spent executing the `run_time_response` function, as well as the time spent specifically on each resolved matrix (A, B, C, D) inside that function, considering different solvers.

The tests were based on the example case, which involves a rotor with 6 shaft elements (n), 2 discs and 2 bearings:

```python
import ross as rs
import numpy as np

i_d = 0
o_d = 0.05
n = 6
L = [0.25 for _ in range(n)]

shaft_elem = [
    rs.ShaftElement(
        l,
        i_d,
        o_d,
        material=steel,
        shear_effects=True,
        rotary_inertia=True,
        gyroscopic=True,
    )
    for l in L
]

disk0 = rs.DiskElement.from_geometry(
    n=2, material=steel, width=0.07, i_d=0.05, o_d=0.28
)
disk1 = rs.DiskElement.from_geometry(
    n=4, material=steel, width=0.07, i_d=0.05, o_d=0.28
)

stfx = 1e6
stfy = 0.8e6
bearing0 = rs.BearingElement(0, kxx=stfx, kyy=stfy, cxx=0)
bearing1 = rs.BearingElement(6, kxx=stfx, kyy=stfy, cxx=0)

rotor = rs.Rotor(shaft_elem, [disk0, disk1], [bearing0, bearing1])

speed = 500.0
size = 1000
node = 3
probe1 = (3, 0)

t = np.linspace(0, 10, size)

F = np.zeros((size, rotor.ndof))
F[:, 4 * node] = 10 * np.cos(2 * t)
F[:, 4 * node + 1] = 10 * np.sin(2 * t)

response = rotor.run_time_response(speed, F, t)
```

In the tests, I also evaluated the execution time by increasing the number of shaft elements to see the impact on each solver. Therefore, I performed tests with n = 6, 60, 600.  Due to my computer's memory limitations, I couldn't perform tests with even larger matrices.

The obtained results are shown below:

|  | la.solve | las.spsolve | las.spsolve ⬌ la.solve | pypardiso.spsolve | pypardiso.spsolve ⬌ la.solve |
| --- | --- | --- | --- | --- | --- |
| n = 6 | Time (s) | Time (s) | Difference (%) | Time (s) | Difference (%) |
| matrix A | 0.106466 | 0.029040 | -72.72348611 | 0.872114 | 719.1508659 |
| matrix B | 0.001645 | 0.000929 | -43.5425424 | 0.001409 | -14.34990578 |
| matrix C | 0.046659 | 0.003133 | -93.2856756 | 0.005273 | -88.69857589 |
| matrix D | 0.001611 | 0.000803 | -50.15534842 | 0.001426 | -11.52537358 |
| run_time_response() | 0.255683 | 0.076050 | -70.2560745 | 1.120431 | 338.2103023 |

|  | la.solve | las.spsolve | las.spsolve ⬌ la.solve | pypardiso.spsolve | pypardiso.spsolve ⬌ la.solve |
| --- | --- | --- | --- | --- | --- |
| n = 60 | Time (s) | Time (s) | Difference (%) | Time (s) | Difference (%) |
| matrix A | 0.042566 | 0.026940 | -36.7089742 | 0.050017 | 17.50590924 |
| matrix B | 0.021948 | 0.008853 | -59.66347303 | 0.019942 | -9.140976341 |
| matrix C | 0.037821 | 0.039090 | 3.35363163 | 0.066180 | 74.98203412 |
| matrix D | 0.010803 | 0.020782 | 92.36626059 | 0.020052 | 85.60866879 |
| run_time_response() | 0.768463 | 0.749327 | -2.490164138 | 0.899261 | 17.02066748 |

|  | la.solve | las.spsolve | las.spsolve ⬌ la.solve | pypardiso.spsolve | pypardiso.spsolve ⬌ la.solve | pypardiso.spsolve ⬌ las.spsolve   |
| --- | --- | --- | --- | --- | --- | --- |
| n = 600 | Time (s) | Time (s) | Difference (%) | Time (s) | Difference (%) | Difference (%) |
| matrix A | 1.525152 | 1.471298 | -3.531072298 | 0.811309 | -46.80471262 | -44.85759441 |
| matrix B | 0.676644 | 0.612068 | -9.543559839 | 0.280629 | -58.52638255 | -54.15073004 |
| matrix C | 1.854077 | 1.700611 | -8.277241723 | 1.177627 | -36.48449367 | -30.75272972 |
| matrix D | 0.872791 | 0.798451 | -8.517512249 | 0.515031 | -40.99034843 | -35.4962321 |
| run_time_response() | 476.871500 | 483.746300 | 1.44164623 | 459.702500 | -3.600340972 | -4.97033259 |

Examining the results, the PyPardiso solver was faster only for the largest system (n = 600) in contrast to the other solvers. However, this performance did not notably impact the overall execution time of the `run_time_response` function. Subsequently, I ran the tests again for n = 600 using cProfile, and the results below reveal that a substantial portion of the code execution time was attributed to the `lsim` function.

[image]

---

[image]

---

[image]

raphaeltimbo, 2024-02-01:
Hi @jguarato!
Thank you for your analysis and clear explanation.
So ROSS would not benefit from using pypardiso since our models typically will not have a huge number of elements. I will close this issue.

Thank you again @jguarato !

closed 2024-02-01 by raphaeltimbo

jguarato, 2025-05-02:
I conducted a series of tests on the `THDCylindrical` class by running its example from the ROSS library and modifying the linear solver used. Two solvers were compared:
```python
# NumPy solver
solve = lambda A, b: numpy.linalg.solve(A, b).ravel()

# Pypardiso solver
solve  = lambda A, b: pypardiso.spsolve(scipy.sparse.csr_matrix(A), b)
```
In the tests, I varied the number of circumferential elements (10, 100, 250, 500, 750, 1000).

First, I performed a mesh dependency analysis focusing on the maximum pressure. The results showed negligible variation beyond approximately 300 circumferential elements:
[image]

Next, I compared the simulation time for different numbers of circumferential elements. As shown below, for smaller meshes (fewer than ~300 elements), the NumPy solver was faster. However, as the number of elements increased, the Pypardiso solver became more efficient than NumPy:
[image]