petrobras/ross #1059: Z direction using probes

ipereiraumich, 2024-05-16:
Hi,

From the documentation, one can notice that 6 dof shaft elements and bearings are implemented but I could not find a way to access the system response in the Z direction. From what I could understand, if one decides to use the probes, it is only possible to access the response in the x and y directions.  Is there any way to access time response in the axial direction as well?

Note that, for the angular misalignment, it is important to have access to the time response in the z direction because this fault is known as generating axial loads as you can see [here](https://skftechnicalsupport.zendesk.com/hc/en-us/article_attachments/360042513054).

raphaeltimbo, 2024-05-23:
Hi @ipereiraumich !
Currently we do not have access to the z direction response, but we are planning to implement this.
We are still trying to figure it out what would be the best way to do it regarding the API.
One option would be to add the the `Probe` class an option to do something like this:

```python
ax_probe = rs.Probe(node=0, angle="axial")
```

We also need to check how we are going to modify the code to select the correct dof in each analysis function.

We will have some discussions about this and will update this issue with the progress.

raphaeltimbo, 2024-08-22:
Closed in #1075.

closed 2024-08-22 by raphaeltimbo