petrobras/ross #1061: run_static() and attribute bearing_forces_nodal do not work with a 6dof rotor

ipereiraumich, 2024-05-22:
The code below defines the same rotor with 4 and 6 dof. Note that  run_static() and attribute bearing_forces_nodal does not work with a 6dof rotor

import numpy as np
import ross as rs
from ross.defects import *
from ross.defects.misalignment import misalignment_flex_parallel_example
from ross.units import Q_
import matplotlib.pyplot as plt
from numpy.fft import fft, ifft
from scipy.signal import welch, csd

"""Create example rotor with a given number of elements."""
steel2 = rs.Material(name="Steel", rho=7850, E=2.17e11, G_s=81.2e9)

i_d = 0      # inner diameter
o_d = 0.019  # outer diameter
n   = 33     # number of nodes

L = np.array(
        [0  ,  25,  64, 104, 124, 143, 175, 207, 239, 271,
         303, 335, 345, 355, 380, 408, 436, 466, 496, 526,
         556, 586, 614, 647, 657, 667, 702, 737, 772, 807,
         842, 862, 881, 914]
         )*1.5/ 1000

L = [L[i] - L[i - 1] for i in range(1, len(L))]

shaft_elem = [
    rs.ShaftElement6DoF(
        material=steel2,
        L=l,
        idl=i_d,
        odl=o_d,
        idr=i_d,
        odr=o_d,
        alpha=8.0501,
        beta=1.0e-5,
        rotary_inertia=True,
        shear_effects=True,
    )
    for l in L
]

Id = 0.003844540885417
Ip = 0.007513248437500

disk0 = rs.DiskElement6DoF(n=12, m=2.6375, Id=Id, Ip=Ip)
disk1 = rs.DiskElement6DoF(n=24, m=2.6375, Id=Id, Ip=Ip)

kxx1 = 8.55e5
kyy1 = 5.21e7
kzz = 0
cxx1 = 27.4
cyy1 = 2.505
czz = 0
kxx2 = 1.19e6
kyy2 = 7.02e8
cxx2 = 50.4
cyy2 = 100.4553

bearing0 = rs.BearingElement6DoF(
    n=4, kxx=kxx1, kyy=kyy1, cxx=cxx1, cyy=cyy1, kzz=kzz, czz=czz
)
bearing1 = rs.BearingElement6DoF(
    n=31, kxx=kxx2, kyy=kyy2, cxx=cxx2, cyy=cyy2, kzz=kzz, czz=czz
)

rotor_6dof = rs.Rotor(shaft_elem, [disk0, disk1], [bearing0, bearing1])

shaft_elem2 = [
    rs.ShaftElement(
        material=steel2,
        L=l,
        idl=i_d,
        odl=o_d,
        idr=i_d,
        odr=o_d,
        #alpha=8.0501,
        #beta=1.0e-5,
        rotary_inertia=True,
        shear_effects=True,
    )
    for l in L
]

Id = 0.003844540885417
Ip = 0.007513248437500

disk0_2 = rs.DiskElement(n=12, m=2.6375, Id=Id, Ip=Ip)
disk1_2 = rs.DiskElement(n=24, m=2.6375, Id=Id, Ip=Ip)

kxx1 = 8.55e5
kyy1 = 5.21e7

cxx1 = 27.4
cyy1 = 2.505

kxx2 = 1.19e6
kyy2 = 7.02e8
cxx2 = 50.4
cyy2 = 100.4553

bearing0_2 = rs.BearingElement(
    n=4, kxx=kxx1, kyy=kyy1, cxx=cxx1, cyy=cyy1
)
bearing1_2 = rs.BearingElement(
    n=31, kxx=kxx2, kyy=kyy2, cxx=cxx2, cyy=cyy2
)

rotor_4dof      = rs.Rotor(shaft_elem2, [disk0_2, disk1_2], [bearing0_2, bearing1_2])

static_analysis = rotor_4dof.run_static()

fs              = static_analysis.bearing_forces

print("Bearing forces - nodes")
print(fs)

print("Bearing forces - nodes - rotor class")
print(rotor_4dof.bearing_forces_nodal)

fs_6dof  = rotor_6dof.run_static()

print("Bearing forces - nodes - 6dof")
print(fs_6dof.bearing_forces_nodal)

print("Bearing forces - nodes - 6dof")
print(rotor_6dof.bearing_forces_nodal)`

jguarato, 2024-05-23:
Hi @ipereiraumich!

What type of error did you get exactly?
Instead of `print(fs_6dof.bearing_forces_nodal)`, wouldn't it be `print(fs_6dof.bearing_forces)`?

ipereiraumich, 2024-05-23:
Error when trying to run_static() for 6dof rotor:
[image]

Error when trying to use attribute bearing_forces_nodal for 6dof system
[image]

Note that both errors do not happen when the same rotor is defined as a 4dof system. The code showing the error is the one above.

raphaeltimbo, 2024-05-23:
Hi @ipereiraumich !
Are you using the ross version installed with pip or the latest code version that we have in the main branch of our repository?
We believe that this is already fixed, but we have not yet published an update release on pypi with this fix.

closed 2024-08-22 by raphaeltimbo