petrobras/ross #1100: BallBearing has no n_balls attribute error

hserdogan94, 2024-08-20:
Hello,

I encountered an issue when I was trying to look for Campbell plot. When I run the code I get an attribute error with development version of ROSS. Despite I defined n_balls attributes.

{
	"name": "AttributeError",
	"message": "'BallBearingElement' object has no attribute 'n_balls'",
	"stack": "---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[17], line 3
      1 samples = 31
      2 speed_range = np.linspace(0, 1900, samples)
----> 3 camp = rotor3.run_campbell(speed_range,6,frequency_type='wn')
      4 camp.plot(harmonics=[1, 2], frequency_units=\"Hz\")

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\units.py:194, in check_units.<locals>.inner(*args, **kwargs)
    191     else:
    192         base_unit_kwargs[k] = v
--> 194 return func(*base_unit_args, **base_unit_kwargs)

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\rotor_assembly.py:2458, in Rotor.run_campbell(self, speed_range, frequencies, frequency_type)
   2456 modal_results = {}
   2457 for i, w in enumerate(speed_range):
-> 2458     modal = self.run_modal(speed=w, num_modes=2 * frequencies)
   2459     modal_results[w] = modal
   2461     if frequency_type == \"wd\":

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\units.py:194, in check_units.<locals>.inner(*args, **kwargs)
    191     else:
    192         base_unit_kwargs[k] = v
--> 194 return func(*base_unit_args, **base_unit_kwargs)

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\rotor_assembly.py:673, in Rotor.run_modal(self, speed, num_modes, sparse, synchronous)
    613 @check_units
    614 def run_modal(self, speed, num_modes=12, sparse=True, synchronous=False):
    615     \"\"\"Run modal analysis.
    616 
    617     Method to calculate eigenvalues and eigvectors for a given rotor system.
   (...)
    671     >>> fig = modal.plot_mode_2d(mode2)
    672     \"\"\"
--> 673     evalues, evectors = self._eigen(
    674         speed, num_modes=num_modes, sparse=sparse, synchronous=synchronous
    675     )
    676     wn_len = num_modes // 2
    677     wn = (np.absolute(evalues))[:wn_len]

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\units.py:194, in check_units.<locals>.inner(*args, **kwargs)
    191     else:
    192         base_unit_kwargs[k] = v
--> 194 return func(*base_unit_args, **base_unit_kwargs)

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\rotor_assembly.py:1367, in Rotor.eigen(self, speed, num_modes, frequency, sorted, A, sparse, synchronous)
   1319 \"\"\"Calculate eigenvalues and eigenvectors.
   1320 
   1321 This method will return the eigenvalues and eigenvectors of the
   (...)
   1364 91.796...
   1365 \"\"\"
   1366 if A is None:
-> 1367     A = self.A(speed=speed, frequency=frequency, synchronous=synchronous)
   1369 if synchronous:
   1370     evalues, evectors = la.eig(A)

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\rotor_assembly.py:1159, in Rotor.A(self, speed, frequency, synchronous)
   1154 I = np.eye(self.ndof)
   1156 # fmt: off
   1157 A = np.vstack(
   1158     [np.hstack([Z, I]),
-> 1159      np.hstack([la.solve(-self.M(frequency, synchronous=synchronous), self.K(frequency)), la.solve(-self.M(frequency,synchronous=synchronous), (self.C(frequency) + self.G() * speed))])])
   1160 # fmt: on
   1162 return A

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\rotor_assembly.py:1010, in Rotor.K(self, frequency, ignore)
    985 \"\"\"Stiffness matrix for an instance of a rotor.
    986 
    987 Parameters
   (...)
   1006        [ 6.,  0.,  0.,  1.]])
   1007 \"\"\"
   1008 K0 = np.zeros((self.ndof, self.ndof))
-> 1010 elements = sorted(set(self.elements) - set(ignore), key=self.elements.index)
   1012 for elm in elements:
   1013     dofs = list(elm.dof_global_index.values())

File c:\\Users\\sssho\\AppData\\Local\\Programs\\Python\\Python312\\Lib\\site-packages\\ross\\bearing_seal_element.py:420, in BearingElement._eq_(self, other)
    418 init_args_comparison = []
    419 for arg in init_args:
--> 420     comparison = getattr(self, arg) == getattr(other, arg)
    421     try:
    422         comparison = all(comparison)

AttributeError: 'BallBearingElement' object has no attribute 'n_balls'"
}

hserdogan94, 2024-08-20:
The ball bearings are defined as  below;

ballbearingR5 = rs.BallBearingElement(
    n=24,
    n_balls= 19,
    d_balls=8.731e-3,
    fs= 710,
    alpha=5*pi/36,
    tag='R5',
    )

#ballbearingR5

ballbearingR4 = rs.BallBearingElement(
    n=25,
    n_balls= 19,
    d_balls=8.731e-3,
    fs= 710,
    alpha=5*pi/36,
    tag='R4',
    )

jguarato, 2024-08-22:
Hi @hserdogan94 ,

This error is related to a method that checks for equality between `BearingElement` instances. 
I have addressed this issue in PR #1102. Thanks for bringing it to our attention.

PR #1102 (2024-09-19): Fix error in equality method for bearing elements
Closes #1100

---

An **AttributeError** occurs when running the analysis of a rotor with ball bearings due to an issue in the `__eq__` method of the `BearingElement` class. This method checks for equality between elements but fails when comparing instances of the `BallBearingElement` class. The error arises because some arguments used during the initialization of a `BallBearingElement` instance (such as `n_balls`, `d_balls`, `fs`, and `alpha`) are not stored as attributes of the object, leading to a mismatch in the comparison.

To resolve this [... cut]
closed 2024-09-19 by raphaeltimbo