petrobras/ross #1101: Upgrade to numpy 2.0

raphaeltimbo, 2024-08-22:
We need to upgrade ross to work with numpy 2.0.
This means that we will have to change our test matrix so that we will only support python >= 3.9.

jguarato, 2024-08-26:
### For the record:

It appears that in NumPy 2.0.0, the representation `repr` of NumPy scalars changed, causing conflicts when running pytest.

See: [NumPy 2.0.0 Release Notes](https://numpy.org/devdocs/release/2.0.0-notes.html#representation-of-numpy-scalars-changed)

### Possible solutions:

1. Create a _conftest.py_ file with a pytest hook to modify options for doctests:

      ```python
      def pytest_configure(config):
          np.set_printoptions(legacy='1.25')
      ```
      This will ensure compatibility with the previous NumPy behavior.

2. Update all doctests to align with the new NumPy behavior.

raphaeltimbo, 2025-03-06:
Since we are always using numpy > 2, I think we should remove the pytest configuration with the np legacy option.