petrobras/ross #1152: that's invalide " autoscale" in page for plot_rotor suca

Pierre-zhu, 2025-03-02:
the code as following, but that's invalide "autoscale' in browser "edge".

`fig_rotor=rotor.plot_rotor(nodes=node_increment)
fig_rotor.show()`

[image]

jguarato, 2025-03-06:
Hi @Pierre-zhu!

In `plot_rotor(),` a scale ratio of 1.5 is fixed for the x-axis in relation to the y-axis. I believe this may be causing the problem you faced. So, you can remove the value ​​defined for this attribute of the figure object or you can also define the value ​​you prefer as shown below:

```python
rotor.plot_rotor(nodes=node_increment, xaxis=dict(scaleanchor=None, scaleratio=None))
```

Pierre-zhu, 2025-03-06:
OK，thanks,I will try as your guide.

Pierre-zhu, 2025-03-09:
that's work well!

closed 2025-03-09 by Pierre-zhu