petrobras/ross #1153: Inaccurate Unbalance Response for rotors with cross coupled journal bearings

PeterSinelnikov, 2025-03-07:
I was running some tests to compare unbalance response results with other commercial software that I've been using for years and have found that it produces results that are very close to the real-world unbalance responses.

ROSS library produced results that were very close to the commercial counterpart, but as soon as I introduce cross-coupled bearing coefficients Cxy and Cyx, the results are not what I would expect.

[image]

Compare to this output:

[image]

In both cases I use identical setup, the same rotor model, the same bearing coefficients, supports, point masses and so on.

Bearing coefficients I'm using are as follows:

[image]

Could anybody comment on this? It fills like there's something how it accounts for Cxy and Cyx that throws it off.

jguarato, 2025-03-14:
Hi @PeterSinelnikov!

Could you share your script so we can take a closer look at the issue?

Based on the information and results you provided, we think this may be related to the damping of the system. Are you considering proportional damping in your model? By default in ROSS, if the values ​​for the `alpha` and `beta` coefficients are not specified when instantiating the `ShaftElement`, the proportional damping is set to zero.

We ran some tests to check this issue and obtained for the same model:

- Without proportional damping:
[image]

- With proportional damping:
[image]

PeterSinelnikov, 2025-03-15:
@jguarato Thank you for taking the time to look into this. I appreciate it.

Here's my code:
```
import time
from pathlib import Path

import numpy as np

import ross as rs
from ross import Probe, Q_

shaft_file = Path("gen_rotor.xls")
shaft = rs.ShaftElement.from_table(
    file=shaft_file, sheet_type="Model", sheet_name="Model"
)
start_time = time.time()
bearing1 = rs.BearingElement(n=6, n_link=51, kxx=469218000, kxy=-152207000, kyx=-1328630000, kyy=2408770000,
                             cxx=1477630, cxy=-2479570, cyx=-2479570, cyy=12537000)
bearing2 = rs.BearingElement(n=40, n_link=52, kxx=469218000, kxy=-152207000, kyx=-1328630000, kyy=2408770000,
                             cxx=1477630, cxy=-2479570, cyx=-2479570, cyy=12537000)

support1 = rs.BearingElement(n=51, kxx=1.75E9, kyy=1.75E9, cxx=709930, cxy=0, cyx=0, cyy=709930)
support2 = rs.BearingElement(n=52, kxx=1.75E9, kyy=1.75E9, cxx=709930, cxy=0, cyx=0, cyy=709930)

bearings = [bearing1, bearing2, support1, support2]

pm0 = rs.PointMass(n=51, m=5000)
pm1 = rs.PointMass(n=52, m=5000)
pointmasses = [pm0, pm1]

rotor = rs.Rotor(shaft, bearing_elements=bearings, point_mass_elements=pointmasses)

probe_angle = 45
probe1 = Probe(6, Q_(probe_angle, "deg"), tag="T1 X")
probe2 = Probe(40, Q_(probe_angle, "deg"), tag="T2 X")

unbalance_response = rotor.run_unbalance_response([15, 31], [100, 100],  # kg.m
                                                  [Q_(0, "deg"), Q_(0, "deg")],
                                                  np.arange(start=400 * np.pi / 30, stop=3600 * np.pi / 30,
                                                            step=20 * np.pi / 30)
                                                  )
print(f"Elapsed time: {time.time() - start_time:.2f} s")
bode = unbalance_response.plot(probe=[probe1, probe2], phase_units="degrees", amplitude_units='µm pkpk',
                                          frequency_units="rpm")

bode.show()
```

Rotor file is attached.

[gen_rotor.xls](https://github.com/user-attachments/files/19256669/gen_rotor.xls)

Please let me know if there's anything else you may need to troubleshoot this.

PeterSinelnikov, 2025-10-01:
This is great news! Thank you!

closed 2025-10-01 by PeterSinelnikov