petrobras/ross #1186: bearing.n passed as a float and not as an integer

ndohme, 2025-07-22:
Hello,

When creating a rotor, I encountered the problem that a bearing node (_bearing.n_) was passed as a float and not as an integer, which meant that it could not be processed correctly in the following lines (file: _rotor_assembly.py_). I then converted the variable into an integer as a workaround so that my rotor creation worked. However, I don't know where the float came from and if it could lead to further errors elsewhere in the code.

```
for bearing in aux_rotor.bearing_elements:
    print(type(bearing.n))
    bearing.n = int(bearing.n)
    bearing_force_nodal[f"node_{bearing.n:d}"] = reaction_forces[bearing.n]
    bearing_force_tag[f"{bearing.tag}"] = reaction_forces[bearing.n]
```

I am using the ‘latest’ version of ROSS, which I installed from github with `pip install git+https://github.com/petrobras/ross.git`. If you have any questions, I can provide further details.

jguarato, 2025-07-25:
The `Rotor` class does not support adding or modifying bearings directly after instantiation, as doing so may lead to internal inconsistencies. However, you can create a new rotor by copying the existing elements and including the updated bearings:

## Adding a new bearing
```python
from copy import deepcopy as copy

# Create a new bearing
new_brg = rs.BearingElement(4, kxx=1e6, kyy=0.8e6, cxx=0, tag="New Bearing")

# Add the new bearing to the existing bearing list
new_bearing_elements = [new_brg, *rotor1.bearing_elements]

# Create a new rotor with the updated bearing list
rotor2 = rs.Rotor(
    shaft_elements=copy(rotor1.shaft_elements),
    disk_elements=copy(rotor1.disk_elements),
    bearing_elements=new_bearing_elements
)

rotor2.plot_rotor()
```

## Modifying a bearing node

```python
# Deep copy the bearings to avoid altering the original rotor
modified_bearings = copy(rotor2.bearing_elements)

# Find and update the desired bearing
for bearing in modified_bearings:
    if bearing.n == 4:
        bearing.n = 6
        break

# Create a new rotor with the modified bearing node
rotor3 = rs.Rotor(
    shaft_elements=copy(rotor2.shaft_elements),
    disk_elements=copy(rotor2.disk_elements),
    bearing_elements=modified_bearings
)

rotor3.plot_rotor()
```

ndohme, 2025-07-28:
Thanks! The inconsistencies were what I was afraid about as well. Your solution looks promising and much nicer than mine, I will probably use it in the final set-up.

Regarding my original problem: Would it be wise to make a change in the instantiation code of the `BearingElement` class to force the node `n` to be an integer, or issue a warning if it is not? Maybe someone else will make the same mistake I did.

Any other questions I have are not related to this issue, so it can be closed from my side.